LaserWeb/LaserWeb4 #427: RasPi service not starting

danowar, 2017-10-18:
I installed the server on a Pi Zero W following the steps on the Wiki.

I am able to manually start the server with: `node server.js`, and all seems to be working properly.

I cannot get it to run as a service with systemd however.  When checking the status with: `sudo systemctl status lw.comm-server`, here is what I get.

● lw.comm-server.service - LaserWeb4 server
   Loaded: loaded (/etc/systemd/system/lw.comm-server.service; enabled)
   Active: failed (Result: start-limit) since Tue 2017-10-17 20:11:33 CDT; 9min ago
  Process: 772 ExecStart=/usr/bin/node /home/pi/lw.comm-server/server.js (code=exited, status=203/EXEC)
 Main PID: 772 (code=exited, status=203/EXEC)

Oct 17 20:11:33 LaserPI systemd[1]: lw.comm-server.service holdoff time over, scheduling restart.
Oct 17 20:11:33 LaserPI systemd[1]: Stopping LaserWeb4 server...
Oct 17 20:11:33 LaserPI systemd[1]: Starting LaserWeb4 server...
Oct 17 20:11:33 LaserPI systemd[1]: lw.comm-server.service start request repeated too quickly, refusing to start.
Oct 17 20:11:33 LaserPI systemd[1]: Failed to start LaserWeb4 server.
Oct 17 20:11:33 LaserPI systemd[1]: Unit lw.comm-server.service entered failed state.

I am not very familiar with configuring systemd services, so I apologize ahead of time if it is something simple that should be common knowledge.  I have searched for an answer without much luck.

Thanks

cprezzi, 2017-10-18:
I am not familliar with Pi Zero. Do you use the same Raspbian Jessie on the Zero?

danowar, 2017-10-18:
Yes, the Raspbian images are the same.

As stated, I can manually start the server, do while waiting for ideas on this issue, I decided to play around some. 

I ran a quick greyscale raster job, and the Pi Zero was not able to keep up with the massive amount of gcode used for rastering.

I was afraid this may be the case, so I have decided to pick up another Pi3 to run the server. Because of this, I am no longer going to persue thus issue.

hayttle, 2018-10-18:
Laserweb4 server happens the same mistake