LaserWeb/LaserWeb4 #428: Windows 8 startup crash

domenic-d, 2017-10-18:
Starting LaserWeb from Windows 8 results in the following exception error.
Tested multiple LW versions:
4.0990-127
4.0.989-115
4.0.734-96

[image]

Any suggestions?

Protoneer, 2017-10-18:
Did you install it as local administrator?

cprezzi, 2017-10-18:
That looks like a problem with the network port. Do you have any firewall or other software that blocks local port 8000?

You could try to use a different port by adding "WEB_PORT=1234" to a ".env" file.

domenic-d, 2017-10-19:
@cprezzi Can you describe how to add this to a .env file?

cprezzi, 2017-10-19:
Just make a new text file in the installation folder with the line `WEB_PORT=xxxx` (replace xxxx with the port number) and save it as `.env` (nothing in front of the dot!). Such files are usualy not shown in the file explorer of windows. To see it, you need to change the explorer settings to show hidden files.

domenic-d, 2017-10-19:
Thx Claudio. I will advise how this work.

cprezzi, 2017-10-20:
This entry changes the port on which the lw.comm-server listen for websocket connections from the frontend. If that works (no more error on start), you also need to tell the frontend to use the new port by changing the "server connection" on the comms tab.

domenic-d, 2017-10-24:
@cprezzi We are still trying to work with the user having this problem. Unfortunately they are not comfortable working at this level, creating hidden files etc.
Users in educational environments have many restrictions placed on their networks, and they are not always skilled with performing the tasks suggested above.

It may be worth considering making this easier for users when they are faced with this problem. Perhaps a way to change this setting without manually editing files.

tbfleming, 2017-10-24:
Oops. Network restrictions. If it's a US public school, then there's probably a county-level network administrator who put the restrictions in place. He or she will have to get involved. I suspect other countries have similar positions.

tbfleming, 2017-10-24:
I know an admin from a nearby county. He locks systems down hard to protect against students who know too much and against school personal who don't know enough.

domenic-d, 2017-10-24:
That's exactly what we are seeing. This issue has come many times already. It's only recently that we got some information to pass on to help diagnose the issue.

cprezzi, 2017-10-25:
There isn't much we can do, if a system is locked down that much. The communication between the frontend an the server (even if on the same machine) works with websockets and this needs an IP port to be open for the server listening to connections from clients.

@domenic-d If there is any port available to run the server on, you could compile a special version for this client, that uses a different default port. You can define the default port in config.h (of lw.comm-server).

domenic-d, 2017-11-14:
@cprezzi I made this change and re-built but the only available port shows was still 8000.

Is there anywhere else I need to change the port for this to work?

domenic-d, 2017-11-15:
@jorgerobles Thanks for the heads up on making the change to the right branch!
This has fixed the issue.
I would encourage the developers to consider making it easier for users to change their server port should they run into this same issue.

closed 2017-11-15 by domenic-d

djjeanb, 2018-01-11:
for windows users how to make a file without filename:
1, make txt file with your settings etc.
2, rename txt file to ".env." notice the dot at the end !!!!!
windows will automatically remove dit at the end making it a nameless ENV file