LaserWeb/LaserWeb4 #429: Support for the Pinecone Robotics XPlotter and other dumb machines

brian000, 2017-10-22:
I recently received my [XPlotter ](https://www.kickstarter.com/projects/pineconerobotics/xplotter-desktop-plotter-laser-cutter-and-engraver)from [Pinecone Robotics](https://www.pinecone.ai/)' recent Kickstarter and I would love to be able to use it with LaserWeb.

As far as I can tell, this thing is _extremely_ dumb.  It doesn't respond to anything but very simple gcode.  I was able to cut/engrave a couple images using the software that Pinecone provided, however that software is Not Good.

I was able to sniff my serial line while running their "official" cutting software and saw nothing but simple GCode.  If I connect directly to the port with Putty I can manually type in the gcode and things move around/laser turns on/etc so I know the machine is in fact working correctly.  No matter what I tell it to do, even on invalid commands it responds with `ok`.

Here's the beginning of using Pinecone's "GUI" software, I simply connect, turn the laser on, and cut a line.

```<20171018200849.523 SYS>
COM is open
<20171018200849.523 SYS>
Baud rate 115200
<20171018200849.523 SYS>
RTS on
<20171018200849.523 SYS>
DTR on
<20171018200849.523 SYS>
Data bits=8, Stop bits=1, Parity=None
<20171018200849.523 SYS>
Set chars: Eof=0x00, Error=0x00, Break=0x00, Event=0x00, Xon=0x11, Xoff=0x13
<20171018200849.523 SYS>
Handflow: ControlHandShake=(DTR_CONTROL), FlowReplace=(TRANSMIT_TOGGLE, RTS_CONTROL, XOFF_CONTINUE), XonLimit=2048, XoffLimit=512
<20171018200849.523 SYS>
Set timeouts: ReadInterval=0, ReadTotalTimeoutMultiplier=0, ReadTotalTimeoutConstant=0, WriteTotalTimeoutMultiplier=0, WriteTotalTimeoutConstant=0
<20171018200849.601 RX>
PR [len=2]
<20171018200849.601 RX>
<LF>
<20171018200849.851 TX>
G28<LF>
<20171018200851.742 RX>
ok<LF>ok<LF>
<20171018200901.304 TX>
M11<LF>					
<20171018200901.304 RX>
ok<LF>
<20171018200904.760 TX>
M4<LF>   				
<20171018200904.760 RX>
ok<LF>
<20171018200906.323 TX>
M5<LF>					
<20171018200906.323 RX>
ok<LF>
<20171018200911.041 TX>
M5<LF>
<20171018200911.041 RX>
ok<LF>
<20171018200911.104 TX>
G1 X52.16 Y0.0 F4500<LF>
<20171018200911.104 RX>
ok<LF>
<20171018200912.167 TX>
M5<LF>
<20171018200912.167 RX>
ok<LF>
<20171018200912.230 TX>
G1 X52.16 Y49.22 F4500<LF>
<20171018200912.230 RX>
ok<LF>
<20171018200923.200 TX>
M4<LF>
<20171018200923.200 RX>
ok<LF>
<20171018200924.153 TX>
M5<LF>
<20171018200924.153 RX>
ok<LF>
```

After running LaserWeb with `DEBUG_LEVEL=3` and `VERBOSE_LEVEL=3` set, I get this on the console.  It's sitting there trying to ask the machine what firmware it's running.  Since it's not responding with anything but `ok` LW gives up.  
```webpack: Compiled successfully.
App disconnected! (id=0)
App connected! (id=0)
INFO: Requesting Server Config
INFO: Connecting to USB,COM6,115200
Sent: ctrl-x
INFO: Connected to COM6 at 115200
Recv: PR
Sent: version
Sent: {fb:n}
Recv: ok
Recv: ok
Recv: ok
Sent: M115
Recv: ok
No supported firmware detected. Closing port COM6
INFO: Port closed
```

Basically, would it be possible to configure LW for "really stupid machines?"  Is there a hidden environmental variable somewhere that can be set to disable firmware detection and make it simple barf out gcode over the serial line?

jorgerobles, 2017-10-22:
LW need some kind of acknowledge to know the machine is ready to be used :(

jorgerobles, 2017-10-22:
From XPlotter site:
> The mainboard we used is ATmega2560, Arduino compatible. We will open source the operation interface for users to extend their own applications on the basis of the XPlotter.

So you can flash a MarlinKimbra :)

brian000, 2017-10-22:
Right, but I’m reluctant to flash anything as they haven’t released the source so I can revert if necessary.  I’d rather have a functioning laser cutter instead of a brick.

jorgerobles, 2017-10-22:
That's true. Ask them the firm .hex first :)

cprezzi, 2017-10-23:
We **need** some sort of firmware identification string because different firmwares need different protocolls. Whithout that, it will not work. 
If they just could add any firmware string that they send out either on connect or after a command like M115 (marlin style), then we probably can support their firmware too.

tbfleming, 2017-10-23:
My guess: it doesn't have an ident string because they're trying to hide something...

brian000, 2017-10-23:
It /does/ belch out `PR` upon opening a connection to it.  Pretty sure that stands for Pinecone Robotics, the device manufacturer.

cprezzi, 2017-10-25:
I guess they just use some grbl or marlin fork, but don't want to tell. 
If I should add support for that machine, I need a real ident string with firmware version (for later compatibility issues) and the spec of their communication protocol. For example: How is laser power set? With S value in G1/2/3 moves or separate commands like M3 Sx? What commands do jogging, homing, set zero....

brian000, 2017-10-25:
They released the [GCode spec](https://docs.wixstatic.com/ugd/c2aaab_a673c9ac3aad4a0eac5e48dd74d1dc52.pdf) this morning.

Looks pretty standard to me.  G0 and G1 for linear move, G2/3 arc, etc.  No way to identify the machine though other than the `PR` upon connection.

jorgerobles, 2017-10-25:
Just ask them to implement M115 if not available. You can say you have
several machines and want to know which is the one.

tbfleming, 2017-10-25:
> M3 S100 ; Set laser brightness to 100
> M3 S255 ; Set laser brightness to 255

The path cam can do this, but the rasterizer can't.

cprezzi, 2017-10-26:
The way they use G92 looks like Marlin, but they probably removed some functions like M115.
Setting the laser power with M3 Sx means you will get stop-n-go raster engraving. This is not worth implementing into LaserWeb. I would suggest to flash MarlinKimbra instead.

jorgerobles, 2017-10-26:
Ca use avrdude to backup the binary
http://forum.arduino.cc/index.php?topic=6150.0

jorgerobles, 2017-10-30:
Sorry but better ask Pinecone an .hex for backup and then try to firm a Marlin Kimbra as formerly suggested.

closed 2017-10-30 by jorgerobles