LaserWeb/LaserWeb4 #430: Desktop file is missing Categories= entry

probonopd, 2017-10-22:
The desktop file is lacking a `Categories=` entry in the `*.desktop` file.

According to the [menu spec](https://standards.freedesktop.org/menu-spec/latest/apa.html),

> By including one of the Main Categories in an application's desktop entry file, the application will be ensured that it will show up in a section of the application menu dedicated to this category. If multiple Main Categories are included in a single desktop entry file, the entry may appear more than once in the menu. 

Hence, please add at least one of the following in the `Categories=` key.

Main Category | Description | Notes
-- | -- | --
AudioVideo | Application for presenting, creating, or processing multimedia (audio/video) |  
Audio | An audio application | Desktop entry must include AudioVideo as well
Video | A video application | Desktop entry must include AudioVideo as well
Development | An application for development |  
Education | Educational software |  
Game | A game |  
Graphics | Application for viewing, creating, or processing graphics |  
Network | Network application such as a web browser |  
Office | An office type application |  
Science | Scientific software |  
Settings | Settings applications | Entries may appear in a separate menu or as part of a 		"Control Center"
System | System application, "System Tools" such as say a log viewer or network monitor |  
Utility | Small utility application, "Accessories"

In addition, you could specify one or more from the longer list of [Additional Categories](https://standards.freedesktop.org/menu-spec/latest/apas02.html).

Please test the result with `desktop-file-validate` and make sure it passes.

__electron-builder offers native support for this since [v19.22.1](https://github.com/electron-userland/electron-builder/releases/tag/v19.22.1).__. Please see the `category` key at https://www.electron.build/configuration/linux-other.

Reference: https://github.com/AppImage/AppImageHub/issues/2

References:
https://travis-ci.org/AppImage/appimage.github.io/builds/291127331#L702
https://github.com/AppImage/appimage.github.io/issues/2

jorgerobles, 2017-10-22:
Thanks! You are welcome to collaborate on LW. Please do a pull request :)
checkout lw.comm-server on electron builder branch.

<https://standards.freedesktop.org/menu-spec/latest/apa.html>,

<https://github.com/electron-userland/electron-builder/releases/tag/v19.22.1>.*.
configuration/linux-other.

Reference: AppImage/AppImageHub#2

References:
https://travis-ci.org/AppImage/appimage.github.io/builds/291127331#L702
AppImage/appimage.github.io#2

jorgerobles, 2017-10-30:
0f0c145d380fbf4ba76cb61feac488e82d6a17ab

closed 2017-10-30 by jorgerobles