LaserWeb/LaserWeb4 #433: Differentiate between cut and travel speed

nerdbeere, 2017-11-06:
Hey there,

I'm pretty new to laser cutting but have a background in 3d printing where we are used to set different speeds for travel an actual printing.
I wonder why there's no such thing in laserweb4. It would significantly speed up my engraving jobs. Is there any reason why this is not implemented? Am I missing something?

jorgerobles, 2017-11-06:
Thanks, but it was a design decision, as far as I remember isn't it @tbfleming ?

tbfleming, 2017-11-06:
@jorgerobles yep.

nerdbeere, 2017-11-06:
@tbfleming Could you elaborate on why this design decision was made? I would love to understand this a bit better.

tbfleming, 2017-11-06:
I just left a comment on the push which partially explains my position. LW assumes that the firmware controls G0 feed. Most firmware has a setting for it.

edgary777, 2018-03-21:
This is an old issue but I recently came upon this same problem, my solution is not perfect solution but works if your firmware doesn't control G0 feed.
You can edit the LW gcode setting for **TOOL OFF** and add "G1 F???" where "?" is the feed speed, it will change the feed speed only when the laser is off.

nerdbeere, 2018-03-22:
@edgary777 what I didn't know when I opened this PR was that you can disable `burnWhite` (not sure what this does anyway) when using Laser Raster which basically does exactly what I wanted to achieve.

Also, if your input file allows that (vectors), use Laser Cut Fill Path. This is much faster than Laser Raster

closed 2018-05-01 by jorgerobles