LaserWeb/LaserWeb4 #435: Check Size incorrect behaviour

domenic-d, 2017-11-08:
It appears that the 'check size' option only shows the correct x position and y position.

When pressed, the laser travels to the correct x-extremity but transverses the complete y range of the workspace.

cprezzi, 2017-11-08:
The function searches for the smalles and largest x and y values in the gcode. You must have some gcode with such coordinates.

domenic-d, 2017-11-08:
@cprezzi There is no gcode or documents loaded with those proportions.  It is a simple circle shape.
Would a workspace file help you?

cprezzi, 2017-11-08:
I would just need the generated gcode (export), not the whole workspace file. 

The outline function is quite simple and dumb. It just parses the whole gcode for all X.. and Y.. values, independant of the rest on the line. So a G0 X0 Y0 at the end of the gcode would mess up the result.

allstar87, 2017-11-14:
I'm also seeing strange behavior out of Check Size; it moves through the origin regardless of where zero is set or where the artwork is in relation.  As far as I can tell, there are no instances of the string "X0" or "Y0" in the generated G-Code.

Tom-Jans, 2018-01-05:
i have the same thing.  i made some [screenshots](https://imgur.com/a/nWU4Y) screenshots to show what i mean. How it is currently setup it doesn't show me if my cut/engrave is on my material. and that makes the function useless in my opinion.

jorgerobles, 2018-01-05:
Please attach your workspace. Cannot debug without that.

Tom-Jans, 2018-01-05:
here is the workspace file because isn't that big i just embedded it as code.

```{
  "documents": [
    {
      "id": "bbc279c8-adf3-4481-891f-75de59ba72f6",
      "type": "document",
      "name": "Blank_square.svg",
      "mimeType": "",
      "isRoot": true,
      "children": ["25036b25-f535-47ae-8834-bfb89a6a5869"],
      "selected": false,
      "visible": true,
      "transform2d": null,
      "rawPaths": null,
      "strokeColor": null,
      "fillColor": null,
      "dataURL": "",
      "originalPixels": null,
      "originalSize": null
    },
    {
      "id": "25036b25-f535-47ae-8834-bfb89a6a5869",
      "type": "rect",
      "name": "rect: rect1306",
      "mimeType": "",
      "isRoot": false,
      "children": [],
      "selected": true,
      "visible": true,
      "transform2d": [
        1.9040081651012883,
        0,
        0,
        1.9040081651012883,
        299.49811913633573,
        199.4981104513615
      ],
      "rawPaths": [
        [
          0.2635917601947719,
          105.30713116868492,
          105.30515810734353,
          105.30713116868492,
          105.30515810734353,
          0.26359632161126,
          0.26359176019477937,
          0.26359632161126,
          0.2635917601947719,
          105.30713116868492
        ]
      ],
      "strokeColor": [0, 0, 0, 1],
      "fillColor": [0, 0, 0, 0],
      "dataURL": "",
      "originalPixels": null,
      "originalSize": null
    }
  ],
  "operations": [],
  "settings": {"toolFeedUnits": "mm/min"}
}```

jorgerobles, 2018-01-05:
Thanks @Tom-Jans, I've been able to reproduce the issue. Will fix ASAP.

jorgerobles, 2018-01-05:
The issue is related to checking min and max coordinates of generated Gcode, that will lead to 0,0 if Homing Gcode is present to that coords. Possible fixes are picking only the min/max of the documents, or keeping the start/end gcode apart from the documents generated GC.

jorgerobles, 2018-01-05:
Done.. Will be on next binary.

Tom-Jans, 2018-01-05:
that was fast. Thanks will check on next release.

closed 2018-01-31 by jorgerobles