LaserWeb/LaserWeb4 #437: RaspberryPI, no USB connection to ch341 arduino nano with GRBL

Geeklab3D, 2017-11-12:
Hi
I followed the installation instructions on the website and installed the latest Raspian on a new Raspberry Pi 3. I also enabled Serial connection in the Raspberry Pi configuration. The webinterface work, except that the connected arduino nano ATmega 341 with GRBL don't show up in the machine connection. 

In the Terminal dmesg shows me:

usbcore: registered new interface driver ch341
[    3.115097] usbserial: USB Serial support registered for ch341-uart
[    3.115186] ch341 1-1.2:1.0: ch341-uart converter detected
[    3.117234] usb 1-1.2: ch341-uart converter now attached to ttyUSB0

.......
I installed the Arduino IDE and get full access to the nano. I flashed the latest GRBL Version to the nano and still have no connection. 

If I connect my W10 Notebook with Eleksmaker to the nano, everything works normal. 

Any idea, why this is not working? Thank you in advance.
[image]

thystonius, 2017-11-14:
I had this similar sounding issue ("No results found") on the Raspberry Pi and found my issue was with the install of the "SerialPort" module for node.
The new guide has the lines:

```
sudo npm install serialport --unsafe-perm --build-from-source
sudo npm install

```
Which worked for me.

Geeklab3D, 2017-11-14:
Hi thystonius
Thank you for your feedback. I will check that tomorrow.

ofanged1, 2017-12-14:
I'm running into the same problem as Geeklab3D.

$ dmesg |grep ch341
[1824054.331645] usbcore: registered new interface driver ch341
[1824054.331930] usbserial: USB Serial support registered for ch341-uart
[1824054.332108] ch341 1-1.5:1.0: ch341-uart converter detected
[1824054.339545] usb 1-1.5: ch341-uart converter now attached to ttyUSB0
[1833136.016131] ch341-uart ttyUSB0: ch341-uart converter now disconnected from ttyUSB0
[1833136.016270] ch341 1-1.5:1.0: device disconnected
[1835925.591298] ch341 1-1.5:1.0: ch341-uart converter detected
[1835925.597659] usb 1-1.5: ch341-uart converter now attached to ttyUSB0`

$ lsusb 
Bus 001 Device 007: ID 1a86:7523 QinHeng Electronics HL-340 USB-Serial adapter

I did install serialport with the same parameters thystonius recomended

thystonius, 2017-12-14:
Can you describe your setup?
Pi version, Linux distro, version of node

I can try and build a quick test script to ensure that serialport is
installed and working properly.
I'll try and get to this tomorrow.

ofanged1, 2017-12-14:
$ cat /proc/device-tree/model
Raspberry Pi 2 Model B Rev 1.1

$ cat /etc/os-release 
PRETTY_NAME="Raspbian GNU/Linux 8 (jessie)"
NAME="Raspbian GNU/Linux"
VERSION_ID="8"
VERSION="8 (jessie)"

$ node -v
v6.12.2

$ npm -v
3.10.10

Eleksmaker EleksLaser-A5 Pro with Grbl 1.1

thystonius, 2017-12-14:
Try this.  Create a new file in your lw.comm-server folder named
"serialtest.js" (save folder as server.js):

var SerialPort = require('serialport');

var port = new SerialPort('/dev/ttyACM0');

SerialPort.list(function (err, ports) {

        console.log('Ports', ports);

        console.log('Error:', err);

        });

// open errors will be emitted as an error event

port.on('error', function(err) {

  console.log('Error: ', err.message);

});

Then run using:

node serialtest.js

Lets see the output.  I suspect that serialport is still not quite
installed somehow, but this should give a definitive answer.

ofanged1, 2017-12-14:
Tried this
'var SerialPort = require('serialport');
var port = new SerialPort('/dev/ttyUSB0', {
  baudRate: 57600
});

// Open errors will be emitted as an error event
port.on('error', function(err) {
  console.log('Error: ', err.message);
});

port.open(function (err) {
  if (err) {
    return console.log('Error opening port: ', err.message);
  }
  // Do nothing
});'
and got
'$ DEBUG=* node testSerial.js 
Error:  Port is opening'

I'll try yours now

ofanged1, 2017-12-14:
$ DEBUG=* node testSerial1.js 
Ports [ { comName: '/dev/ttyACM0',
    manufacturer: 'Prusa_Research__prusa3d.com_',
    serialNumber: 'Prusa_Research__prusa3d.com__Original_Prusa_i3_MK2_CZPX0717X003XK12123',
    pnpId: 'usb-Prusa_Research__prusa3d.com__Original_Prusa_i3_MK2_CZPX0717X003XK12123-if00',
    vendorId: '0x2c99',
    productId: '0x0001' },
  { comName: '/dev/ttyAMA0',
    manufacturer: undefined,
    serialNumber: undefined,
    pnpId: undefined,
    vendorId: undefined,
    productId: undefined },
  { comName: '/dev/ttyUSB0',
    manufacturer: '1a86',
    serialNumber: '1a86_USB2.0-Serial',
    pnpId: 'usb-1a86_USB2.0-Serial-if00-port0',
    vendorId: '0x1a86',
    productId: '0x7523' } ]
Error: null

thystonius, 2017-12-14:
Hmmm, so I'm certainly wrong about "serialport" installation.
Just to confirm when you go to the Comms tab, then click on Server
Connection you should have the ip address of your Pi there.  It should NOT
be 'localhost'.

ofanged1, 2017-12-14:
Hrm, my Pi is localhost. I have localhost there... localhost:8000 to be exact

ofanged1, 2017-12-14:
and from the Pi
$ curl -I localhost:8000
HTTP/1.1 200 OK

What should it be if not localhost?

thystonius, 2017-12-14:
This is the address used by the LaserWeb javascript app in your browser to
connect to the Pi.
Just to confirm you are not running Chromium from the pi (which to my
knowledge is not supported for memory & performance reasons), but running
another browser on another machine to connect to LaserWeb on the Pi.
Give it a quick try changing this to the ip address of pi (still using port
8000) and see what happens.

ofanged1, 2017-12-14:
Using the wireless adapter ip worked. Very weird! But apparently, I'll need to redo the fw deployment, as I guess it didnt actually happen.

Firmware grbl 0.9j detected
 Grbl version too old -> YOU MUST INSTALL AT LEAST GRBL 1.1e

Thanks for your help, Nathan. Any idea why we have to use the IP of the machine? I have a DNS name resolving to that machine locally, could I use that so I dont lose the address?

thystonius, 2017-12-14:
The client (your browser) needs to connect to the Pi and the server
connection is also from the client to the Pi.  This connection is not the
Pi to itself, but the browser to the Pi.  So any client side dns lookup
should work.  It just cannot be only local to the pi.
Good luck with the Grbl update :-).

cprezzi, 2017-12-14:
Did you point the "server ip" to the raspi IP on port 8000 and open the "server connection"?

Geeklab3D, 2017-12-14:
Hi. I complete reinstalled the pi from scratch. I get now the serial connection to the arduino but having the GRBL to old issue. I will try to flash the arduino with the new Firmware this weekend.

jorgerobles, 2018-05-01:
Is this solved?

gibsonsystems, 2018-05-23:
Same issue here.  Starting over.  I am using an Eleksmaker A3 Pro 2500mw PWM laser with a Raspberry Pi 3+.  I connect to the web fine but the machine connection dropdown is not populated.

ellisgl, 2018-09-12:
Strange - I show the USB ports ttyUSB0 and ttyAMA0, I can connect to ttyUSB0 @ 115200 from the CLI, but when I try to connect via the web interface, nothing! I'm using serialport@6.2.2 

/var/log/daemon.log:
Sep 11 20:38:08 raspberrypi lw.comm-server[5548]: INFO: Connecting to USB,/dev/ttyUSB0,115200
Sep 11 20:38:08 raspberrypi lw.comm-server[5548]: /home/pi/lw.comm-server/server.js:291
Sep 11 20:38:08 raspberrypi lw.comm-server[5548]:                     parser: serialport.parsers.readline('\n'),
Sep 11 20:38:08 raspberrypi lw.comm-server[5548]:                                                ^
Sep 11 20:38:08 raspberrypi lw.comm-server[5548]: TypeError: serialport.parsers.readline is not a function
Sep 11 20:38:08 raspberrypi lw.comm-server[5548]:     at Socket.<anonymous> (/home/pi/lw.comm-server/server.js:291:48)
Sep 11 20:38:08 raspberrypi lw.comm-server[5548]:     at Socket.emit (events.js:182:13)
Sep 11 20:38:08 raspberrypi lw.comm-server[5548]:     at /home/pi/lw.comm-server/node_modules/socket.io/lib/socket.js:503:12
Sep 11 20:38:08 raspberrypi lw.comm-server[5548]:     at process._tickCallback (internal/process/next_tick.js:61:11)
Sep 11 20:38:08 raspberrypi systemd[1]: lw.comm-server.service: Main process exited, code=exited, status=1/FAILURE
Sep 11 20:38:08 raspberrypi systemd[1]: lw.comm-server.service: Unit entered failed state.
Sep 11 20:38:08 raspberrypi systemd[1]: lw.comm-server.service: Failed with result 'exit-code'.
Sep 11 20:38:08 raspberrypi systemd[1]: lw.comm-server.service: Service hold-off time over, scheduling restart.
Sep 11 20:38:08 raspberrypi systemd[1]: Stopped LaserWeb4 server.
Sep 11 20:38:08 raspberrypi systemd[1]: Started LaserWeb4 server.

After some playing, it looks like the newest version of serial port has some major changes, promises and stuff..

cprezzi, 2018-09-12:
The lw.comm-server needs serialport version 4.x not 6.x! And you should donnect the controller via USB cable, not RX/TX to the Raspberry Pi.

Also important, as @thystonius wrote: When you use a Raspberry Pi as the server, you must set the IP of the Pi (+:8000) in the Server field of LaserWeb, not localhost:8000!

ellisgl, 2018-09-12:
Well I couldn't get 4.x to compile under Node 10, so 6.x works under 10.

I might take a look making changes to server.js to get 6.x working tonight.

cprezzi, 2018-09-13:
As written in the wiki, we need NodeJS V6 ;)

cprezzi, 2018-09-13:
A port to nodejs 10.x with serialport 6.x would be very much appreciated.

ellisgl, 2018-09-13:
I know it says V6, but I hate being that far back. =) I have a fork going for lw.comm - but I've only done code styling at the moment. I do have much personal time at the moment, so it might be a couple days before I get something working.

cprezzi, 2018-09-13:
Please don't do code styling / reformating! Thant makes it very complicated to review a pull request (find the relevant changes) and will therefore not confirmed/accepted.

thystonius, 2018-09-13:
I'd be willing to help with that V10 update.  Do you have any other
specific packages that you do / do not want updated?  Any other things to
include / avoid?

ellisgl, 2018-09-13:
The only reason why I did code style formatting was so I can better read it, so it's more for me. Once I find the way of getting 6.2.2. working, I can pull the original back and place the fix. Also seeing the size of server.js, it might be better to restructure the code like this: https://blog.risingstack.com/node-hero-node-js-project-structure-tutorial/

cprezzi, 2018-09-14:
@thystonius I don't expect too many problems with the update. Just the servialport lib changed from 4 to 6 so usage must be adjusted, but the update should be worth the work because the new version should be faster and more stable.

I turn of coding: I'm not a fan of too many dependencies, so if possible I do avoid them and rather copy a function over or do it myself.
Because I was the only maintainer of lw.comm-server over the past 2-3 years, I prefer not to change the structure of the code. Shure, it's not very readable, but I'm perfectly in it ;)

cprezzi, 2018-09-14:
@ellisgl Server.js is just a big state machine and MVC does not make sence for me. What theoretically would make sense is separating the different supported firmwares and connections to separate files, but it would increase the time to implement new features to all versions.

thystonius, 2018-09-14:
Great.  I’ll start a fork. I’m also a believer in only breaking one thing at a time so I’ll only change what is necessary to update then you can review.

My only limitation is that I will only have access to grbl based laser, so that will be the limitations of my ability to test.

ellisgl, 2018-09-14:
@cprezzi Should have been more specific. I believe that server.js could be broken down in to modules. I've started to play with it in my fork (experimental (I typo'd the name though) branch). Basically breaking down communication and machine stuff into their own modules and DRYing things up.

cprezzi, 2018-09-17:
@ellisgl I have merged your pull request. Looks good so far. 
Did you also try to compile the electron app? Hope we did not break compatibility.

cprezzi, 2018-09-17:
@thystonius Could you test the new version?

ellisgl, 2018-09-17:
No I didn't test the electron app. Just threw it on my PI and it worked.

cprezzi, 2018-09-17:
Ok, but compiling the electron app is broken now. 
Probably the LW frontend (and all of it's dependencies) also needs to be updated to Node v10. 

@tbfleming What do you think? Should the LW frontend be able to compile on Node v10?

thystonius, 2018-09-17:
I ran into the same thing.  Server was easy, but it broke Electron app so I was trying to get that working.

ellisgl, 2018-09-18:
Where's the code for the Electron app.

cprezzi, 2018-09-18:
It's on lw.comm-server branch electron_bundler (see package.json script nightlywindows).

ellisgl, 2018-09-18:
I'll take a look tonight. It looks like it should be mostly the same
changes.

ellisgl, 2018-09-18:
https://github.com/ellisgl/lw.comm-server/tree/electron_bundler --- I get a blank screen. How do I get the debug tools up?

ellisgl, 2018-09-19:
I figure out how to get dev tools up.
Things I've noticed:
Server is loading up on 8000 and Electron is trying to hit WS on 8080
Changed to 8080 on the server config, and get this:
ERROR:  webServer error:/sockjs-node/info?t=1537321840028 :  Not Found 

This could be from something else, since I was trying to update all the packages. 

Has anyone tried compiling the Electron app from the latest masters?

cprezzi, 2018-09-19:
I don't belief that there is something wrong with the ports, as I am able to compile the electron app under node v6. The electron app serves two ethernet ports, one for the frontend website and one for websocket communication.

ellisgl, 2018-09-19:
Ill dig in a bit more tonight. I'll try just updating the electron version
to 2.0.x and see what happens using the old serial module.

ellisgl, 2018-10-22:
I did a couple pull request to fix the issue over a month ago (On this and the Electron App) and I haven't heard anything, just checking to make sure all is good @cprezzi .

thystonius, 2018-10-23:
Ellis,
I literally just now pulled your commit and was going to do some testing.
I would have done this right after your commit, but just got back from a
long trip from my EU office and couldn't do anything.   I really look
forward to this update.
So everyone is aware, I do think that this will affect the install
instructions on rasp-pi so I'm going to focus my testing in that area.

Nathan

ellisgl, 2018-10-23:
Awesome. I was a little bit worried.

thystonius, 2018-10-23:
Ok, so I did some testing and your Serial Port updates seem to work on my
grbl and smoothie based machines.
In conjunction to that I've made a pull request for the NodeJS v10 update,
but Ellis's changes for Serial Port should be merged first.  My changes do
not include his changes so it should be easier to see what updates were for
each of the two pieces.

I was able to run the electron build and it did build and run, but it could
use some more eyes.  Also, I put this in the comments but, there was also
an update to the .travis.yml file for LaserWeb4 so that will need to be
updated to make the electron app work.

Lastly, there will need to be some updates to some of the install docs to
reflect this version update.  I volunteer to do those, but I'm not very
excited to start that until there is a level of comfort with this update.

cprezzi, 2018-10-24:
Did you also test if the Elektron App could still be compiled? This ist 90% of usage!

cprezzi, 2018-10-24:
Ok, I will check the pull requests the next few days (I'm very busy at the moment).

vang1804, 2018-11-18:
Hi everyone,

Any new regarding the issue? I have tried all the combinations above.

I pull the latest and install serial-port 4.0.6 ... no luck
I pulled the serialport-6.2.2 branch and install ... also no luck..

```
cat /proc/device-tree/model
Raspberry Pi 3 Model B Rev 1.2

cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 9 (stretch)"
NAME="Raspbian GNU/Linux"

root@lw4:/opt/lw.comm-server# npm -v 
3.10.10
root@lw4:/opt/lw.comm-server# node -v
v6.14.4

node  serialtest.js 
Ports [ { comName: '/dev/ttyAMA0',
    manufacturer: undefined,
    serialNumber: undefined,
    pnpId: undefined,
    vendorId: undefined,
    productId: undefined },
  { comName: '/dev/ttyUSB0',
    manufacturer: '1a86',
    serialNumber: '1a86_USB2.0-Serial',
    pnpId: 'usb-1a86_USB2.0-Serial-if00-port0',
    vendorId: '0x1a86',
    productId: '0x7523' } ]

```

any ideas?

ellisgl, 2018-11-18:
Check which version of serial port is installed. Also check develop tools
for errors.

vang1804, 2018-11-18:
I am using serialport@4.0.6.
I am also trying to enable debug on the about page.. but the Toggle Devtools button is not there...

ellisgl, 2018-11-18:
Use serialport@6.2.2 instead.

ellisgl, 2018-11-20:
@vang1804 Did that work for you?

vang1804, 2018-11-20:
Yes, it seem to work with 6.2.2 and the main brunch. I didn't have to time to really test it, but seems that it connected to the GRBL.

ellisgl, 2020-07-26:
Can we close this?

filka96, 2021-03-16:
I have the same problem. I found a reason and fix
https://github.com/LaserWeb/lw.comm-server/issues/91

closed 2021-09-27 by cprezzi

t-richter2, 2022-02-04:
@filka96 Hi, I#m still running into this problem. Can you clarify how the fix works?
Don't have a clue about node.js config and how to apply this.
Many thanks

easytarget, 2022-02-05:
@t-richter2 How did you obtain LaserWeb?
- What platform are you installing lw.comm-server on? A Pi?
- Are you building it from source? if so; where did you get the instructions? 

The discussions and solutions above are for people who download the source code and build that using npm/nodeJS. And these issues are now fixed in the latest LW versions. This is a closed issue.

Please start a new issue, and include info about what you downloaded, from where, and what you then did to install.

t-richter2, 2022-02-05:
> Please start a new issue, and include info about what you downloaded, from where, and what you then did to install.
@easytarget Sorry, me stoopid ;-) [I did just that](https://github.com/LaserWeb/lw.comm-server/issues/101)