LaserWeb/LaserWeb4 #438: LaserWeb laser problems

NoobCNC, 2017-11-20:
Hi,
i have a problem with LaserWeb4. After installing the binaries and connecting the software with my chinese laser engraver i did a few tests and everything was fine. The engraver is controlled by an arduino nano with nano_grbl-edge-1.1e firmware.The next day when i tried to do some engraving i saw that my 5.5W laser doesnt burn anything.I did an EEPROM reset, reflash the firmware but nothing.Steppers work fine, laser fire when it should but no burning.I know the laser is good because if i try to engrave a gcode file generated with 3dpBurner Image2Gcode using Grbl Controller, the laser is burning the wood.Ive played with the settings in LaserWeb but nothing.Does anybody had this problem and found a solution?
I really need some help with this.
Thank you.

DevZTH, 2017-11-24:
Go to settings -> Gcode
in TOOL ON write M3
in TOOL OFF write M5
also check PWM MAX S VALUE it's must be acorded cnc grbl settings in default grbl settings it's 1000

jorgerobles, 2017-11-24:
@DevZTH Thanks for answering!.. @NoobCNC sorry but this issues are better solved on https://plus.google.com/communities/115879488566665599508, Github is for LW bugs.. not configuration.

closed 2017-11-24 by jorgerobles