LaserWeb/LaserWeb4 #439: Laserweb crashes Smoothieboard during jogging

tayken, 2017-12-04:
We are trying to use LaserWeb to run our laser cutter with Smoothieboard. But every time we try to jog it, it crashes the Smoothieboard and the COM port disappears until we unplug and replug. Seen in both Linux (Ubuntu 16.04) and MacOS. Terminal output is attached. If there is more info to provide, please let me know.

[laserweb.txt](https://github.com/LaserWeb/LaserWeb4/files/1527463/laserweb.txt)

jorgerobles, 2017-12-04:
Got three Smoothies and none of them crashes doing Jogging (I should know by now), do you have another board to check is not faulty?
Also, have you checked latest CNC firmware from SmoothieWare?

Thanks!

tayken, 2017-12-04:
That is the weird part for us. We updated the firmware on June 3rd and gave up on it for a while. What is the version you are using at the moment? Wish the firmware came with an actual version number instead of latest.

We will try to update the firmware and run some new tests.

jorgerobles, 2017-12-04:
Latest firm i've got is a month ago or so.
Have you tried other software to determine is only related with Laserweb?

tayken, 2017-12-04:
I used Pronterface to jog the machine successfully and Inkscape gcode addons for some tests. Both were able to run the machine without any problem. Only with Laserweb we see that the COM port suddenly disappearing which causes the problems.

jorgerobles, 2017-12-04:
@cprezzi, any idea?

cojarbi, 2017-12-04:
Ive had those issues when any other program that autoconnects to the port is running at the same time as LW

thenetimp, 2017-12-04:
@jorgerobles,  both @tayken and I have tried using different software on both Mac and Linux and we are able to jog the motors without any problem, and no disconnects from the com ports.  I don't know when we last updated the firmware, as @tayken said it may have been June, so we'll try to update it this weekend and see if we can get it to work.

tayken, 2017-12-04:
@cojarbi, there is no other program running at the time other than LaserWeb that can connect to the COM port on the Ubuntu machine.

tayken, 2017-12-30:
We tried the firmware-latest.bin file today and it solved the communication issue. Rastering is problematic but we will try and see if firmware-cnc-latest.bin solves the issue.

Closing the issue.

closed 2017-12-30 by tayken