LaserWeb/LaserWeb4 #440: Shell to launch an engraving

lhanneus, 2017-12-09:
Hi, I don't find a forum to ask this question, so I hope someone will help me or point me a place where I can get some help.
I've a fully working laserweb4 working on a Raspberry connected to a DIY laser engraver (in GRBL).

Is their a way to launch an engraving process of a specific .gcode file **with a shell command** ?

jorgerobles, 2017-12-09:
Currently not. Dou you want to run a prepped .gcode?

lhanneus, 2017-12-09:
Yes.

The idea will be to launch the engraving of specific prepped .gcode with a GPIO connected to a switch.

jorgerobles, 2017-12-09:
@cprezzi?

cprezzi, 2017-12-11:
At the moment the gcode is run by the frontend and just cached in ram on the server (not in a file). So there is no option to run (or rerun) a specific gcode file from the server only. 
You could start a job from a PC and pause it immediately, then resume it from a different frontend (for example on a smartphone).
An option would be to write a separate client that runs on the pi and uses the lw.comm-server websocket API. There you could select the file to run a use a hardware pin to start the job.

lhanneus, 2017-12-12:
Do you think I can run in parallel another python program ? like the "simple_stream.py" from the grbl library : https://github.com/grbl/grbl/blob/master/doc/script/simple_stream.py

Then I just have to modify the serial port, point the correct .gcode file and add some line to launch it when there is an action on a GPIO.

Or it isn't possible ?

cprezzi, 2017-12-12:
Nope. USB ports cannot be shared. Only one process can open a certain port at the same time. 

Did you know that GRBL supports hardware buttons for abort/hold/resume? It should be possible to place a GRBL pause command (!) in the start gcode so the machine waits for the resume button to really start the job.

lhanneus, 2017-12-12:
The resume won't help for this project.

Just to be sure, if I don't press the "connect to serial" in the laserweb interface, it doesn't use the USB port. So it won't be "share" ? So another python program could use that USB port ?

Maybe to explain a litlle more the project :
* The idea is to install the laser engraver in a kind of public space. They have access to 4 different patterns (4 .gcode files) that they could choose (with an external GPIO) and it launched directly the engraving.
* And at other moments of the day, there is a workshop, where peoples understand what they are doing and are using the Laserweb interface where they design their files, upload their svg and ...

cprezzi, 2017-12-13:
That's Correct. If you don't connect the machine from LW4, you can use another application to connect and send data.

lhanneus, 2017-12-15:
Ok, I tested the idea.
It works.

Thanks all for your help.

closed 2017-12-15 by lhanneus