LaserWeb/LaserWeb4 #442: SVG issues.

FabCreator, 2017-12-16:
Hi,

It would seem some SVGs are not being read by laserweb for some reason. 
Below are a selection of files generated from,

http://boxdesigner.frag-den-spatz.de/
and http://www.festi.info/boxes.py/

Some lines work but others are completely ignored by laser web.

I have attached some example files.

File loaded
[image]

After generating g code
[image]

[Example files.zip](https://github.com/LaserWeb/LaserWeb4/files/1564892/Example.files.zip)

jorgerobles, 2017-12-16:
Boxes.py is known for generate ugly svg with collinear shapes. Lw library
clipperjs does not handle collinear well.
 The only way for now is cleaning double paths on editors like AI, and even
there is a challenging task.

f0x52, 2018-01-14:
I made this by hand, and the straight lines of the pentagon aren't being lasered either.
[penta.zip](https://github.com/LaserWeb/LaserWeb4/files/1629747/penta.zip)

f0x52, 2018-01-14:
[image]

tbfleming, 2018-01-14:
@f0x52 please use https://plus.google.com/communities/115879488566665599508 for support questions.

tbfleming, 2018-01-14:
Locking this thread so it doesn't attract more SVG support questions. @FabCreator hopefully you'll still be able to post here since you're a member.