LaserWeb/LaserWeb4 #447: Need a reset button on GUI

radiolee, 2018-01-16:
1, I used android pro mini board and Ch340 usb2serial board with grbl 1.1f, if i connect to Laserweb, I need reset the pro mini manualy.
2. I'm using STM32 port of grbl 1.1f now https://github.com/usbcnc/grbl/ it's a USB Virtual Com Port. I created an .env file with RESET_ON_CONNECT=1, and now can connect the board. But, after I generate the g-code, run the job, xyz still stay 0.00mm, Queued lines is decrease, but no move. I try to send ctrl-x; reset command but no luck. I touch the same issues in cncjs, but it's working ok after I click reset.

So, how can I send a reset to grbl by manual?

tbfleming, 2018-01-16:
There's a keyboard shortcut. Press ctrl+x to send ctrl+x to the board.

cprezzi, 2018-01-16:
Don't forget configure the grbl settings correctly (see https://cncpro.yurl.ch/documentation/initial-configuration/65-grbl-lpc-1-1e) and load the "Generic GRBL Machine" Profile in LW4.

radiolee, 2018-01-16:
thanks for help, I'll check it  later.

closed 2018-01-17 by radiolee

radiolee, 2018-01-17:
@cprezzi @tbfleming Thanks all.  It's working send ctrl+x, after setting grbl LW4 works normally. In the guide https://cncpro.yurl.ch/documentation/initial-configuration/65-grbl-lpc-1-1e I found this line:
$10=0      ;send work coordinates in statusReport (needed for LW4!)
is it mean LW4 only accecpt WPos report? I think this is the point of my issues when I leave $10=1, that's report MPos. But I found this https://github.com/LaserWeb/deprecated-LaserWeb3/pull/172 by @cprezzi in LW3 status report can fallback from WPos to MPos, so I was not care about it.

cprezzi, 2018-01-17:
@radiolee Yes, LW4 needs WPos for the DRO and pointer position.

radiolee, 2018-01-17:
@cprezzi thanks very much to find the root issues.