LaserWeb/LaserWeb4 #448: SVG rendering issue NOT related to zero-area closed paths

MrSurly, 2018-01-16:
### The short version

Changing a single path point coordinate from `191.516` to `191.5165` causes that _entire path_ to **not** be rendered as GCODE.  Yes, by changing the coordinate by 1/10000th of a mm.

### The long version

#### Working correctly:

* Workspace: `tm_good.json`
* SVG: `tm_good.svg`

Screenshot, area highlighted with red circle:
[image]

#### **Not** working correctly:

* Workspace: `tm_bad.json`
* SVG: `tm_bad.svg`

Screenshot, area highlighted with red circle:
[image]

### The actual difference

Only difference between `tm_good.svg` and `tm_bad.svg` (other than the filename metadata) is seen below:
```diff
237c237
<        d="m 191.5165,90.917 2.11984,-2.425725 2.47851,-1.099802 -2.13625,-3.299409 -3.71997,8.087217 z"
---
>        d="m 191.516,90.917 2.11984,-2.425725 2.47851,-1.099802 -2.13625,-3.299409 -3.71997,8.087217 z"
```

In case it's not obvious, the first coordinate as been changed from `191.5165` to `191.516`.  I believe this change should not mean the difference between rendering and not rendering as gcode.

-----

All files: [LW_issue.tar.gz](https://github.com/LaserWeb/LaserWeb4/files/1636424/LW_issue.tar.gz)

tbfleming, 2018-01-16:
At first glance, without running it through analysis, it looks like you may have hit a numeric instability. Sometimes the difference between what any polygon algorithm can and can't handle is minuscule. No known polygon algorithm has solved that. The closest are the integer polygon algorithms, which I use. But float to int conversion reintroduces instability.

MrSurly, 2018-01-16:
So, I presume that you're treating all closed paths as polygons (rather than a series of line segments) , and feeding them through a polygon library, because of their potential to be raster filled?

I can only find the term "numerical stability" in reference to convex hull algorithms.

tbfleming, 2018-01-16:
Yes, but also for inside and outside (kerf compensation).

MrSurly, 2018-01-16:
Okay, new info: There are actually two polygons in that position that have the same path.

Object IDs:  `path4290-8` and `path4294-4`

Not sure why two overlapping polygons causes this ...

MrSurly, 2018-01-16:
>Yes, but also for inside and outside (kerf compensation).

That makes sense

tbfleming, 2018-01-16:
> Okay, new info: There are actually two polygons in that position that have the same path.

They cancel out. LW4 uses the even-odd rule for complex polygons and the xor rule for combining multiple polygons.

MrSurly, 2018-01-16:
> They cancel out. LW4 uses the even-odd rule for complex polygons and the xor rule for combining multiple polygons.

Got it.  I'll keep this in mind for any other random SVGs I get.  Thanks.

closed 2018-01-16 by MrSurly