LaserWeb/LaserWeb4 #449: Google Analytics

Juerd, 2018-01-16:
I'm not too happy about user behaviour being tracked without explicit consent or even a warning. Please remove Google Analytics from the software, or at least request explicit consent from users first.

tbfleming, 2018-01-16:
The person who added it killed the analytics account not long after he left the project. I'm not apposed to removing it.

Peetz0r, 2018-01-16:
This is pretty insane for local running software, even if it's webbased. I would love to help the project with suggestions and other feedback, but I'd like to know when I am giving it. If you want to watch me use it, please come and visit IRL. Don't be sneaky!

tbfleming, 2018-01-16:
As far as I know it only collected usage statistics.

jorgerobles, 2018-01-16:
https://github.com/LaserWeb/LaserWeb4/commit/65b2757cddc2d86c94b4e15bedfdb1453b58b877

jorgerobles, 2018-01-16:
Please check this https://github.com/LaserWeb/LaserWeb4-Binaries/issues/7 if anyone can help. Thanks!

closed 2018-01-16 by jorgerobles

cprezzi, 2018-01-17:
I think it's legit that we want to know how many users we have and how ofthen LaserWeb is used. Don't forget this is free software. If you don't like it, take the work to compile your own version - it's open source!

jorgerobles, 2018-01-17:
I agree, but the tracking account was deleted by Peter as far as I know,
anyway.

cojarbi, 2018-01-17:
We can add any other account. Also maybe make those metrics public to the
users, no harm in that.

tbfleming, 2018-01-17:
I am curious what usage is.

cojarbi, 2018-01-17:
If i remember correctly it just accounted the time spent with the app open

tbfleming, 2018-01-17:
This reminds me of the movie Airplane... I meant, I'm curious how many people use it...

cojarbi, 2018-01-17:
Am i the captain?

Those metric showed at that time about +1000 active users daily , but that
was on LW2 and 3

cprezzi, 2018-01-17:
I realy would like to know how many unsers do use LW4 and how often, how long they use it. 
If there are just a few users left, I would stop developing ;)