LaserWeb/LaserWeb4 #450: dxf-parser updated to support Splines

bzuillsmith, 2018-01-23:
Hey, I thought I'd let the LaserWeb team know that Quadratic and Cubic splines are now supported in dxf-parser 0.5.1. This could eventually lead to some kind of support for spline-to-gcode generation. [Three-dxf has an example of how to use this spline data](https://github.com/gdsestimating/three-dxf/blob/be0b3b296476b8def4bc0ab3a57d6a824e407b72/dist/three-dxf.js#L359-L382). It offloads the heavy calculations to Three.js but I know you guys are adept at pulling from other libraries like Three.js for your needs so that shouldn't be too hard ;)

tbfleming, 2018-01-23:
@iceblu3710

iceblu3710, 2018-01-23:
Yay math! I will take a look, thanks.

bzuillsmith, 2018-01-23:
I'm completely new to splines so I'm realizing Three.js doesn't seem to rational curves (I think). There is no way to put knot values into it. So it's not exactly complete. I suppose some limited support is better than nothing, however it leaves potential for slight differences that the user might not notice right away.

tbfleming, 2018-01-23:
Ignore Three.js; it's not in LW4.

bzuillsmith, 2018-01-23:
Right, my point was simply that it's not an already solved problem in Three.js so it'll require coming up with more of the code yourselves, not that three.js is being used in LasterWeb.

iceblu3710, 2018-01-23:
I reverse engineered three.js

iceblu3710, 2018-01-23:
Accidental mobile boop. Sorry.

Was going to say I reverse engineered three.js for curve support so it's at least a useful head start.

paulr4, 2018-03-20:
Hi, I'm investigating LaserWeb for the first time; it looks great but the lack of DXF spline support is a problem for me. Any word on whether this will be implemented soon?

iceblu3710, 2018-03-22:
So splines open up a different can of worms for me. Right now my code supports R12 ASCII DXF files only. R12 converts all splines into polylines when you save in that format as splines where not supported yet back then.

Sooo, I kind of already have spline support? lol. If you share a real world example I can see what I can come up with.

What do you need spline support for if you don't mind me asking?

paulr4, 2018-03-26:
Hi, thanks for the response, the reason I need spline support is that I'm looking to transition from the laser software I currently use, which although nasty and buggy, does support splines.

I've been using it for some time, and have lots of previously cut artwork files, mostly saved in DXF R27, some of which don't open correctly in LaserWeb as they contain splines. People also submit work as DXFs, and imposing restrictions on the DXF options they can use will cause even more of them to screw up than currently do!

I can see there must be many demands on developers' time, and this may not seem the most vital, but speaking for myself at least, it's a problem not having it.

iceblu3710, 2018-03-26:
R27 is very unlikely to happen as I would need to implement literally hundreds of new commands. The parser also only supports R12 entities plus splinters now. Its on its way to R13 but there are about 25 missing entities like DIMENTION, and HATCH. 

R12's view of the world is "everything is a LWPOLYLINE" which makes it incredibly easy to work with. I recommend you try opening your existing files and simply exporting them to R12 to see how well they translate. You can keep the design in R27, just export R12 for printing. I have done this with an AutoCAD Electrical panel drawing and it worked perfectly.

Give it a try and let me know if something is non-convertible.

paulr4, 2018-03-26:
Hi, saving as R12 does seem to allow my designs to import, so I'm pleased about that. Thank you. I'd still like to see spline support one day, even if R27 isn't supported. It seems more pressing than dimensions for example.