LaserWeb/LaserWeb4 #451: Laser intensity

Kriechi, 2018-01-23:
### Follow this steps to open an issue ###

- [ ] If refers software, include `Workspace.json`.  This is available on *CAM tab top, Workspace -> Save*. Prepare your work, save `workspace.json`, rename as `workspace.json.txt` and attach in this issue. *All resources regarding the issue should be attached (SVG files, images, etc.). An issue opened without resources will be automatically closed upon inspection*

- [ ] If refers hardware, include enough images, log files, and optionally videos (or link to a Cloud Drive folder) to inspect.

Kriechi, 2018-01-23:
Reading a bit of source code tells me, that only `cam-gcode-laser-cut.js` makes use of this setting.
I tested it with a "Laser Cut" job - and it does indeed generate correctly.

Is it possible to also make this work with a "Laser Raster" job?

tbfleming, 2018-01-24:
Duplicate of #395

closed 2018-01-31 by jorgerobles