LaserWeb/LaserWeb4 #453: Coordinates not being displayed

gregoryperrin, 2018-01-28:
Hi,

I am able to jog the machine, but on the top left, X and Y always say 0, even while i jog it.

I tested this on two different computers with Linux Mint on it.

[image]

jorgerobles, 2018-01-28:
GRBL $10=1

closed 2018-01-28 by jorgerobles

gregoryperrin, 2018-01-28:
My grbl settings in my board (Eleksmaker Mana SE) with GRBL 1.1f is already setup with $10=1. Are you suggesting this is the problem and I should make it 0 or 2?

gregoryperrin, 2018-01-28:
@jorgerobles

jorgerobles, 2018-01-28:
Yes, it's on the documentation https://cncpro.yurl.ch/documentation/initial-configuration/31-grbl-1-1e you have to set to $10=0