LaserWeb/LaserWeb4 #454: Is telnet connection only for smoothieboard?

radiolee, 2018-01-31:
I am flash a esp-link in an esp-01 fw today, it's for serial-socket transport.
after check the source code seems only support smoothie , am I wrong?

jorgerobles, 2018-01-31:
Currently yes, but that's not a LW concern, as long other vendors/firm could implement the protocol
https://hackaday.com/2015/08/20/hello-ramps-meet-esp8266/

radiolee, 2018-01-31:
thanks. looks websocketserver is the only change for now.

closed 2018-01-31 by radiolee

radiolee, 2018-02-01:
@jorgerobles I emulate a Grbl 1.1f welcome string in serial tool, and LW4 connect to ESP via Telnet successful using esp-link serial-socket brige. LW4 recognize grbl and emulate "ok/r/n" in serial tool by 10ms interval with no issues. The job run to the end of Gcode.
[image]

radiolee, 2018-02-01:
It's really speed up than esp8266  websocket, and more stable.
[image]

jorgerobles, 2018-02-01:
@radiolee That are good news!

cprezzi, 2018-02-01:
Right, Telnet has less overhead compared with websocket, but 9 lines/s is very slow. It's ok for vector cut/engrave, but not usable for raster engraving, where we need something around 50-100 lines/s, which we reach with USB serial. That's why ESP connection is not officialy supported.

radiolee, 2018-02-01:
@cprezzi thanks for your reply. But I tested ch340 usb2ttl with m328 grbl before using the same gcode(c-beam xy test plate) , the total job time is 12min. I'll test a raster file tomorrow via telnet. The websocket arduino fw is slower than socket to send this gcode file, I get little stutters every 5-10sec under test.

radiolee, 2018-02-02:
@cprezzi OK, after change a serial tool "sscom5" I get 50+lines/s with 10ms interval of send "ok" response.
[image]

radiolee, 2018-02-02:
Hi all, finally I got 62lines/sec send the job "c-beam xy test plate" with line by line reply "ok/r/n" response. The transport bridge is esp-01 with esp-link v3.0.14 stable version, serial<->socket on port 23.

[image]

radiolee, 2018-02-05:
I got the top speed 71lines/s when I update the esp-link to https://github.com/jeelabs/esp-link/releases/tag/v3.2.47.alpha
[image]

radiolee, 2018-02-05:
When I do these tests above, I have some doubts about a few points:
1, The "ok" response interval, >=15ms the performance decline linearly, but <=15ms no help. I guess it's some timer in esp-link or sdk.
2, Is it any way to improve the performance? Maybe increase the esp rx buffer but I heard that it's not a good idea.
3, Why arduino platform (websocketserver) got the poor performance, I think "Telnet has less overhead compared with websocket" is not the only reason.
4, As I emulate grbl, maybe I found an issue, click "connect" in LW4 "Machine Connection", serial tool received "version {fb:n} M115", then I send grbl welcome string "Grbl 1.1f ['$' for help]\r\n" , LW4 command window hit " undefinedGrbl 1.1f ['$' for help]"(see screenshoot), if I send grbl welcome string again, it works. 
[image]
5, When I start a big job, like raster job above, it's very slow if chrome is on the front. Works normal if I switch the chrome minimize, or I got the poor performance both in serial data transfer or the pages display in chrome.

cprezzi, 2018-02-05:
@radiolee I don't understand what you try to reach. Just trying to get the fastest transfer speed out of a serial simulator? Or connect a real Grbl board by wlan? 

The max. performance of Grbl 1.1f on a AVR328 over USB serial is around 60-70 lines/s. Don't expect to get more than that if you bridge the communication through WLAN via ESP8266.

cprezzi, 2018-02-05:
ESP-Link itself is a BIG overhead, because it has too many features that are not needed for a simple WLAN<>Serial bridge. The only thing you need is receive one line of gcode (until CR) from telnet and send it to serial and vice versa. The ok response must come from Grbl and not be emulated on ESP8266 because we need to be synchroneous with the machine, not with some piece of software in between.

cprezzi, 2018-02-05:
About 4)
LW supports Grbl, Smoothieware, TinyG and MarlinKimbra and tires to detect what firmware is connected. For this we first wait 1 second to get the Grbl welcome string. If we don't get that string after this second, we send "version" to get the smoothieware version and wait another second for the answer. If still not gettig the expected answer we send "{fb:n}" to get the TingyG  version and after another timeout we try "M115" to get MarlinKimbra version. All theses timeouts can be configured via environment switsches (see config.js).

radiolee, 2018-02-06:
@cprezzi My goal is findout the best way to replace usb vcp in usbcnc https://github.com/usbcnc/grbl
Thanks for your explain, I'll get on with stable communication through WLAN via ESP8266.

cprezzi, 2018-02-06:
I forgot to mention, that we have a "RESET_ON_CONNECT" environment switch for cases where the board is connected via RX/TX only (no DTR) like with ESP8266. This way we send a reset command to the board after the serial connaction is established.

radiolee, 2018-02-07:
@cprezzi Yes, I had set it in config.js