LaserWeb/LaserWeb4 #455: Server error: Writing to COM port (GetOverlappedResult): operation aborted  Machine disconnected

FabCreator, 2018-01-31:
So I am getting this random error, i have been trying to pin down what it could be. Can go days or weeks without it. But then right when you are using some nice material or engrave and item, this happens.

Server error: Writing to COM port (GetOverlappedResult): operation aborted
 Machine disconnected

[image]

[Laserweb-Workspace test.zip](https://github.com/LaserWeb/LaserWeb4/files/1683585/Laserweb-Workspace.test.zip)
[image]
[image]

FabCreator, 2018-01-31:
If i disconnect, and then reconnect, I no longer get the hold state and i can start the job again from the start.
Origin is remembered.
[image]

jorgerobles, 2018-02-01:
@cprezzi

FabCreator, 2018-02-01:
Okay so i have got to the point I can reliably trigger the error but this time it freeze up the smoothie board. so i have to reset it or power cycle.

If i trigger a Limit, X or Y min during a job it will abort as it should. However if i trigger a X or Y max it will spam the information abort X max triggered and then disconnect with multiple errors of the above. Ill get some screen shots.

cprezzi, 2018-02-01:
If I remember correctly, some time ago someone reported a bug in smoothieware that could cause it to hangup. Unfortunately I don't remeber where this was and if it has been fixed.

If you can tell me which versions of smoothieware and LW you used, I can try to replicate this issue.

FabCreator, 2018-02-01:
http://community.fabcreator.com/t/fabkit-mk5-laser-web-settings-and-profiles/115

Here are all the info regarding the set up. Its currently running 4.0989

We were running the lasest CNC firmware, but im reading now that may have caused issues i reverted back the to "stable" https://github.com/Smoothieware/Smoothieware/blob/edge/FirmwareBin/firmware-cnc.bin

FabCreator, 2018-02-01:
By pressing the X or Y max limit switch, i get this.
[image]

FabCreator, 2018-02-01:
[Smoothie board error.txt](https://github.com/LaserWeb/LaserWeb4/files/1685709/Smoothie.board.error.txt)

This is a copy of the text. As you can see it goes mad with the duplications of the error. The smoothie board freezes and sometimes the USB driver corrupts and i have to reinstall the driver or reboot the pc.

FabCreator, 2018-02-01:
Its starting to look like it may be a smoothie thing.

Testing with pronterface can get the same results.
Interestingly when it decides to freak out it reports the X Min was being triggered. When in fact it was the X Max i was pressing. I don't think many people/machines use the max limits. could be why we are discovering it. Be interested to see if you can get the same result. does the cohesion board have max limit stops?

cprezzi, 2018-02-01:
The remix board does have max limit inputs, but I don't have them configured, because I only use one limit switch per axis. Not sure when I have time to connect additional test switches. 

The console log looks like the com3 port is just gone at some time and only gets back after a reset of the smoothie board. That sound exactly like the report i remeber, where the smoothieboard crashed.

cprezzi, 2018-02-01:
The multiple error lines are probably comming from the noder serial library that still tries to send the last data to the non-existing port.

FabCreator, 2018-02-01:
Okay, so i have just experienced a random Hold.

Machine has halted, smoothie is still flashing normally, no mention of a disconnect 
[image]

If i click pause it then says hold.
[image]

Run, and its idle.
[image]

Click abort, nothing happens
[image]

Click pause, back to hold.
[image]

Click abort, Now aborts like normal
[image]

Clear alarm, and back to hold.
[image]

I can now send the ~ to clear the hold.
and back to normal.

What could be triggering the hold? Is their a way i can clear the hold without aborting first?

I think there are either a few things going on or its all connected, not 100% sure

cprezzi, 2018-02-01:
Is this when connected via USB cable or through ESP8266 WiFi?

FabCreator, 2018-02-01:
USB

FabCreator, 2018-02-01:
Shield and beaded

cprezzi, 2018-02-01:
Ok. 

The status display (idle/hold/alarm) is read from the status messages that we query from the controller aby sending "?" (every 200ms or so). It seems the sending of pause, resume and abort is still working in this situation and the board also executes the commands and responds with the corresponding status.

The question is, why the queue stopped sending data. Most likely we are waiting for the next 'ok' that we don't get or missed for what ever reason. One possible reason could be communication problems. Another could be the firmware not sending the ok for some reason or LW not processing a received ok correctly. A pause/resume cycle should usualy kick our queue to continue.

If that doesn't work, you could try to manually send ~ to resume, but im not sure if that manual command gets added to the end of the queue or sent directly to the board. I will check that.

FabCreator, 2018-02-03:
Just tried out the Telnet with smoothie.  So far the random disconnects are not occurring. Have not tried any rastering yet,
https://github.com/Smoothieware/Smoothieware/issues/1264 referring to the weird crashing i was getting when pressing a max limit switch during the job. I cannot force this error with telnet. only USB.

FabCreator, 2018-02-03:
I cannot send a ~ during this random hold as you must first abort the job in order to clear the hold.
With the fact the telnet works i would imagine somthing funny is happening with the USB coms?