LaserWeb/LaserWeb4 #456: Mill inside/outside not working with some DXF files

grantemsley, 2018-02-01:
[front2.dxf.txt](https://github.com/LaserWeb/LaserWeb4/files/1687387/front2.dxf.txt)
[missinglines.json.txt](https://github.com/LaserWeb/LaserWeb4/files/1687388/missinglines.json.txt)

I'm using Fusion 360 to create DXF files.

I tried using the [Dogbone](https://github.com/caseycrogers/Dogbone) plugin to add notches around inside corners.

Laserweb seems to read the file ok, and if I choose laser cut or mill cutfor the operation it follows the paths perfectly.  But Mill Cut Inside and Mill Cut Outside only cut one small part of the design and ignore the rest.

jorgerobles, 2018-02-01:
I suspect that the plugin makes an open shape, common in DxF  so the inside
algorithm cannot resolve the polygon.

closed 2018-03-05 by grantemsley

Timacaro, 2019-03-25:
Good afternoon,
I am having the same problem, so what should I do with that open shape? How do I solve this?

cprezzi, 2019-03-25:
Search the gap in the shape by zooming in on your CAD software close that gap or open the DXF in inkscape and try it there. Also try it without dogbones, because that could create the gaps.