LaserWeb/LaserWeb4 #457: Check Size doesnt turn on laser

nickm324, 2018-03-02:
[Laserweb-Workspace.json.txt](https://github.com/LaserWeb/LaserWeb4/files/1776660/Laserweb-Workspace.json.txt)

Have a diode laser (cheap Chinese) and have the Check size set and the laser test set. Laser Test works fine with the correct power level but when I click Check Size the movement works but the laser never turns on.
Have gone through the settings a dozen times.

tomcosk, 2018-03-04:
I have the same problem. I don't think it depends on laser type (Chinese) as laser test is working.

nickm324, 2018-03-04:
Well I found a work around that is kind of odd. Just before I hit check size I hit Test Laser and it turns on and I can see where its going to cut and the weird this is that when check size is done it actually turns off the laser I dont have to hit laser a second time to turn it off. So it appears that its trying to work just not getting the command to turn it on but does send command to turn it off.

jorgerobles, 2018-03-04:
As seen on manual:
https://cncpro.yurl.ch/documentation/settings/56-gcode/14-gcode

[image]

Generated gcode as

[image]

You should check you have GRBL laser mode with M4 head gcode so all G1 moves fire the laser, too.

closed 2018-03-04 by jorgerobles

nickm324, 2018-03-04:
I have laser mode enabled in Grbl and have M4 set to turn on the laser. There is nothing in the documentation about "M4 head gcode so all G1 moves fire laser". Can you explain what that means? And I dont see any M4 in the Generated code sample you provided.

jorgerobles, 2018-03-04:
It is Grbl, https://github.com/gnea/grbl/wiki/Grbl-v1.1-Laser-Mode.
The gcode pasted is just a debug to show the S (laser power) command.

Anyways, set the check size percent to the max pwm on settings like 10% or
so, and try.

Regards.

nickm324, 2018-03-05:
I have the check size max power set and thats what I am saying the laser doesnt turn on but if I hit test laser  button before I hit check size button the laser will turn on once the check size starts moving and it will turn the last off when its done. So its like Its sending M5 at the end of check size but it doesnt send M4 before starting.

jorgerobles, 2018-03-05:
@cprezzi?

cprezzi, 2018-03-05:
That's correct - Check Size does not activate the laser (but it uses G1 commands). 

Having the laser active during Check Size only makes sense when you can see the laserbeam (like on diode lasers) but is a big risk on powerfull invisible CO2 lasers. 

Therefore you need to consciously activate the laser with the "Laser Test" button before clicking "Check Size". Or even better: define a macro button with "M4S0" to activate the laser with zero power. The power is then set by the Check Size function according to your settings in "Check Sizte Power". (Hint: M4 only works with Grbl!).

nickm324, 2018-03-05:
Ok that makes sense. Although I did state I was using a Diode laser. I will setup the macro. thanks, just wasn't sure if I was doing something wrong or needed to do some additional setup.