LaserWeb/LaserWeb4 #461: Lasersaur Compatibility

adammhaile, 2018-03-29:
On the wiki it says this requires grbl 1.1 or higher, but on the Lasersaur google group someone mentions this works and as far as I can tell, the lasersaur does not use grbl 1.1 as the last firmware update before it was prior to grbl 1.1 being released. So is that 1.1+ more of a soft requirement or is there something special that needs to be done for Lasersaur support?

adammhaile, 2018-03-29:
Oh oh... should've specified. I have an old Lasersaur. v12.08 to be precise, which very much does run grbl. Just not v1.1
Google group thread I mentioned is here: https://groups.google.com/forum/#!topic/lasersaur/TqsIoOx3xV8

adammhaile, 2018-03-30:
Thanks :) I haven't even fired the thing up fully yet... so will be a bit. But once I get comfortable, updating the firmware is first on my list.

closed 2018-04-17 by jorgerobles

cprezzi, 2018-04-18:
The Lasersaur V12 board uses an ATmega328P Chip in DIP package (like old Arduino Uno) but the pinout is not compatible with GRBL 1.1! 
It should be possible to make a rewiring adapter to use a ATmega328P with GRBL 1.1 on this driver board. At least this is what I have planned for our Lasersaur  ;)

adammhaile, 2018-04-18:
@cprezzi  - My Lasersaur board actually just has an Arduino Uno on it. Which I always thought was weird as it takes me like 10 minutes and 5 components to add a 328P "Arduino" to any board I make. But whatever. 
Also very weird that the pinout is not compatible... is Grbl expecting a chip other than the 328P? Even the QFP/QFN version should have the same pins available.
In the end I kind of gave up on LaserWeb for this... instead I'm trying to update to the latest LaserSaur driveboard firmware so that I can use the newest DriveboardApp. Almost have it working...

adammhaile, 2018-04-18:
@cprezzi - It's been tempting to do something like that but I'm still learning the thing and didn't want to screw it up just yet :P Though you said you were going to make a wiring adapter and use Grbl... why not Smoothieboard like you just suggested?
Also, what version of the LaserSaur do you have? Mine is a 12.08