LaserWeb/LaserWeb4 #464: Laserweb _ New UI proposal

marcodallago, 2018-04-04:
Hi! Time to time I come back over here and I share my ideas ahaha

I made a proposal for a new UI design which aim is open dev's mind on possibilities that this new branding could lead. Hope that there are still people working on this really cool app.
This is a WIP, which means that there is a lack of button and function and card measurement are not so precise.
Let me know what you guys think of it!!

[image]
[image]

cojarbi, 2018-04-04:
Marco

I don't mind changing the UI but there a lot to think about before that.

For example the most trivial part being the workflow on the tools section. I here from many users that it's difficult to understand, yet there are many who have spent a lot of time learning it.

FabCreator, 2018-04-23:
Would be great to get a touch screen UI version of laser web again.

dieck, 2018-04-25:
Hi,
I am more of a form-follow-function kind of guy, and I don't really like those dark and black settings for work (or enthusiastic tinkerer) environments.

I can see the appeal, this seems to be in style for a lot of people.

If it is a XOR decision, I'd vote against it.
I am not (yet) used to node.js, or LaserWeb, development, but maybe enhancing the GUI interfaces with template/skins abilities to allow for multiple UIs might be a way to go?

marcodallago, 2018-05-03:
[image]
[image]

It could have a white version too.

Btw this is just a suggestion, not a real practical design. It needs a deep thinking with the design team. 

Box with inside function are useful to me because anyone can adjust them depending on their workflow. Of course it needs some rigid dimension, but plugin like management could be integrated easily with it.

Of course if this is a way to proceed a LOT must be tought about.

My idea on this design is simple. You have a single page in which you can set up your machine. At first the program could ask to set up machine profile. After that, there are 4 main pages that looks like the actual structure of pages. In each one you can hide or show extra cards depending on your workflow. This is tricky because some cards are essential and other not.
The connection to machine is automatic after someone set it up. Moreover, you can choose between different machine from a drop down menu over "machine name".

I don't know if this could be a way to go or not.

jorgerobles, 2018-05-03:
Hi Marco.
Sincerely, I enjoy "light" theme a lot, but technically, choosing white or black is not a great issue, is just a theme driven by css.

I like the content disposition. Seems uncluttered, giving console relevance, something is now somewhat difficult to type and read. That said, is only a component used by expert people, so may not be useful for all the users (until they need it.)

Upper menu eats same space as the current sidebar, but could work, even exposing the  operation database that needs a better location than the current modal.

I suppose Adjust means Settings, or am I mistaken?

cprezzi, 2018-05-03:
I don't see a big difference from the actual design. Ok, there are round buttons instead of sliders and square buttons, and the console is in the left column, but there are many things missing, like z-axis DRO and buttons, single axis functions (dropdown) macro buttons... 

Shure it looks less cluttered when reduced to just a few features, but thats not enough.

tbfleming, 2018-05-03:
@cprezzi I didn't get around to implementing the commands in the drop-down on the existing UI. I suspect it's a task that we could point new devs at as a way for them to get started.

jorgerobles, 2018-05-03:
I find Marco's design just a feeling mockup,

But that said, in my opinion, to do a good design have to:
 * write down the overall design objectives and declare the priorities, for example, I would say: 
     - Objectives: Declutter and give coherence to all the UI
     - Priorities: 1. SAFETY, 2. RESPONSIVENESS,  3. EASE OF USE, and lastly, 4. DESIGN
 * list all the features that have an UI, describing, and then sorting for relevance and studying the interactions.
 * then throwing some questions (should be generic enough to be answered by all the UI parts), taking the priorities on account.
   - What are the priorities of this component?: 1,2,3,4
   - What is it's purpose? control movements of the machine, set the defaults related to the movement, then do helping tasks involving movement
   - Does it communicate its functionality, clearly? It has a label that indicates that function (jog control), but there are buttons that cannot be recognised clearly. 
   - Does it behave as expected? Taking safety first the first priority is Stop the machine. Do I see which is the stop button? is it accesible enough to be clicked instead another button? What happens if I do a mistake and click another button instead?

Then [mock, evaluate, improve]{n} and then approve.

It's a task as daunting as programming 😅

FabCreator, 2018-05-03:
the interface in general is okay, 
[image] 
From the feedback i have received its this area people find confusing.

JustGand, 2019-01-23:
I find highly unlogic to see the "Comms"  menu entry (used once at startup) between "Files" and "Control", it would be much more logic to put it near "Settings"

cprezzi, 2019-01-23:
@JustGand We have more than 3500 Users (according to G+ community) and you will also get used to it.

closed 2019-01-23 by cprezzi

JustGand, 2019-01-23:
@cprezzi I've been one for a few years, but why consider a new UI in that case ?

I could have modified marcodallago and reordered the icons to illustrate my proposal, but i thought it was easy enough to picture ...

JustGand, 2019-01-23:
@cprezzi I've been one for a few years ...

cprezzi, 2019-01-24:
The actual design is oriented on the workflow: 
1. prepare job
2. connect machine
3. run job

Shure there are other ways to do it, but it's not wise to change such things after a few thousand users are used to it for long time.

cprezzi, 2019-01-24:
By the way: The "New UI proposal" was just a discussion initiated by one of the users and the last message in this discussion was almost a year ago.

marcodallago, 2019-01-24:
To be honest the interface is very dense and full of possibilities. This is a plus of course but also could be confusing depending on what type of user is. Of course this program's aim is to give the most complete experience possibile, and is not related to a specific product with a noob type consumer (see very practical interface as easel). That being said everything can be optimised and improved, but It depends on actually how much devs want to work, considering that the UI in this case is secondary to performance since it's not a product for market. Most important is that all this people are doing it for free..... I'm sure they have other works to focus more on.. 
My work was very naif, and so simple. Of course it was on a very early stage, but was also an experiment to me. An easier css restyling at this point could be the easiest option, considering that there is a reason if stuff are in this position after many thoughts and experiences of users and devs. This, btw, requires still many works and time to spend and I don't think that devs are willing to do that for free (correct me if I'm wrong), since is not so necessary. It's not a company, they are not making money with the software, it's understandable. I don't know how the development is proceeding, but in case you want an help in the css restyling @cprezzi I'm here . Still think that Laserweb is really fantastic for what's is offering :D

cprezzi, 2019-01-24:
@marcodallago Thank you for your offer to help with css restyling, but since the actual state is stable (enough) and usable, there are no plans for big changes at the moment (also because most core devs moved to other projects).

However, we are still open for new devs, if somebody is willing to step into and take over some responsibillity.

marcodallago, 2019-01-24:
[image]

@cprezzi this is just an effortless change I made. Of course it is all messed up on the major interface, but the menu and the element subdivision is a nice try to me :) this is just a css and mininor html change. But, with time can be done :D

cojarbi, 2019-01-24:
I will call it a modern facelift , not sure it hurts anyone, but agree with
@cprezzi not to change the layout or major changes at this point

[image: screenshot 2019-01-24 at 16 28 40]
<https://user-images.githubusercontent.com/16123281/51688618-41a08c00-1ff5-11e9-8329-be421d67ee8e.png>