LaserWeb/LaserWeb4 #466: Size issue SVG PNG JPG

Member32, 2018-04-17:
First I am new to laser burning. I was tired of Benbox randomly stopping with the laser on and burning a hole in the wood.
I cannot get a correct sized burn from laserweb4. I tried resizing in Inkscape and Photoshop. Images are 72dpi. When I open image in Laserweb4 size looks fine attached test png shows 50mm width. When I burn it is larger. If I click Raster Opt it changes the size on screen to 28mm wide on screen, but still burns larger (110mm wide).
Pictures print the correct size on paper (from printer).
I have been looing for a week on issues similar to this but have found nothing.
So either this is a real issue or I am doing something wrong,

[Laserweb4-Workspace bad size.json.txt](https://github.com/LaserWeb/LaserWeb4/files/1917844/Laserweb4-Workspace.bad.size.json.txt)

[image]

jorgerobles, 2018-04-17:
Go to settings, file settings and check the ppi box.

cprezzi, 2018-04-17:
And check that you have configured the correct steps/mm for each axis in the Grbl $ settings.
If you click the arrow buttons in LW4 the head should exatly move the distance that is set by the buttons below (0.1, 1, 10, 100mm). If this is not the case, then the steps/mm settings in Grbl are wrong.

Member32, 2018-04-17:
PPI box was 72.
The 10mm jog was going 20mm on both x&y axis.
It took some more searching.. but below are the settings I changed to make 10mm a real 10mm.
$100=160 changed to 80
$101=160 changed to 80 
In Benbox I looked for some similar setting I saw the PPM setting at 320 (160*2) and started with changing the value to 320, once I realized this increased travel made the new value 80.
Hopefully this will help someone else out.
Thank you for your help!!!!