LaserWeb/LaserWeb4 #468: Power hungry on OSX

lucalcin, 2018-04-23:
Hi,

LaserWeb4 in working fine on OSX, but I noticed that there is a very high power requirement even if LW4 is idle.
In my MacBook Pro LW4 uses 25%-80% of the computer resource when idle, and 100% when imports, export or calculate g-code.
I suppose there is some power-consuming routine that can be optimized
Please ask for further info/test
In attachment the task manager that shows LW4 in idle state consuming 63% of the resources.

[image]
[image]

carlosen14, 2018-05-05:
same problem..

lucalcin, 2018-05-24:
No news?
[image]

jorgerobles, 2018-05-24:
Cannot reproduce the issue:

jorgerobles, 2018-05-24:
Of course, LW is OpenGL intensive, I've got discrete video (Nvidia) that could be the fix for the issue.

carlosen14, 2018-05-25:
@jorgerobles what graphic card do you have? I have this specs.. 
[image]

Maybe is the poor graphic card the one that complains and works harder.

carlosen14, 2018-05-25:
@jorgerobles what graphic card do you have? I have this specs..
Maybe is the poor graphic card the one that complains and works harder.

[image]

jorgerobles, 2018-05-25:
It's on the pic :)

Nvidia gtx760.

lucalcin, 2018-05-25:
[image]

jorgerobles, 2018-05-25:
https://github.com/LaserWeb/LaserWeb4/issues/216

Enable settings/application/display cache