LaserWeb/LaserWeb4 #470: "Join Pixels" gives incorrect results with "Verbose GCode" disabled

llamasoft, 2018-04-23:
Using "Join Pixels" without "Verbose GCode" enabled leads to incorrect results.  If you compare files 1, 2, and 3, you'll see that files 1 and 2 are functionally identical, but file 3 will result in... nothing.  In general, the issue is that the first point in any series of burnt points is stripped.  When the image is mostly single points (e.g. when the dithering option is enabled), large numbers of points are stripped.

As best I can tell, the root cause is in [`_command`](https://github.com/LaserWeb/LaserWeb4/blob/dev-es6/src/lib/lw.raster2gcode/raster-to-gcode.js#L254-L282) and [`postProcessing`](https://github.com/LaserWeb/LaserWeb4/blob/dev-es6/src/lib/cam-gcode-raster.js#L63-L130).  
`_command` is stripping all common values between the current and previous command, which results in a bare `S` command that `postProcessing` strips for being invalid.  
One suggested fix would be prevent `_command` from removing `G` elements, another would be to have `postProcessing` prepend a `G1` to bare `S` commands.

**Relevant Files:**
[1_noJoin_Verbose.txt](https://github.com/LaserWeb/LaserWeb4/files/1940318/1_noJoin_Verbose.txt)
[2_Join_Verbose.txt](https://github.com/LaserWeb/LaserWeb4/files/1940319/2_Join_Verbose.txt)
[3_Join_noVerbose.txt](https://github.com/LaserWeb/LaserWeb4/files/1940320/3_Join_noVerbose.txt)
[vertical_bars.png](https://user-images.githubusercontent.com/4632450/39157623-a738b4e8-4729-11e8-8454-123949d58894.png)
[workspace.json.txt](https://github.com/LaserWeb/LaserWeb4/files/1940321/workspace.json.txt)

cprezzi, 2018-04-24:
Also the noJoin gcode looks wrong to me, because there are zero lengs G1 moves. I'm wondering why it even works. Need to investigate some more.

llamasoft, 2018-04-24:
Is zero length G1 invalid?  I assumed it was the only way to burn a point that is the exact size of the laser diameter.  The only thing that isn't being controlled is the duration of the laser pulse before the next G1 occurs.  I'll verify if it actually burns points next time I'm at my local makerspace.

As for why there are zero-width G1: it appears to be a result of [`_getPoint`](https://github.com/LaserWeb/LaserWeb4/blob/dev-es6/src/lib/lw.raster2gcode/raster-to-gcode.js#L340-L345).  The transition from "white->black->white" means that the first white point will have `lastWhite = true` which gives it a positive offset, then the next point (black) has `lastColored = true` which gives a negative offset.  Here's how it manifests in the GCode:

```
; Size       : 0 x 0 mm
; PPI        : x: 127 - y: 127
; PPM        : x: 0.2 - y: 0.2
; Tool diam. : 0.2 mm
; Feed rate  : 4000 mm/min
; Beam range : 0 to 1
; Beam power : 0 to 100 %
; Options    : trimLine, burnWhite, verboseG

G1 F4000

G1 X0.10 Y0.10 S0.0000
G1 X0.10 Y0.10 S1.0000
G1 X0.40 Y0.10 S0.0000
G1 X0.60 Y0.10 S0.0000
G1 X0.80 Y0.10 S0.0000
G1 X1.10 Y0.10 S0.0000  ; Would have been X1.0, but lastWhite = True
G1 X1.10 Y0.10 S1.0000  ; Would have been X1.2, but lastColored = True
G1 X1.40 Y0.10 S0.0000
```

llamasoft, 2018-04-27:
I got to run a few tests today.  The results were as you predicted.  No matter the power, a zero-width move results in nothing.  I even tried a zero-width move (`G1 S1.0000`) followed by dwell (`G4 P10`) and the result was nothing.  It seems like the laser ceases to fire as soon as the G1 command finishes.  
My initial idea for `lastColored`/`lastWhite` hack kind of worked.  It resulted in visible single-point dots, but resulted in incorrect GCode for longer strings of powered points.  I'll play around with it some more and make a pull request if I come up with a workable solution.

closed 2018-05-01 by jorgerobles