LaserWeb/LaserWeb4 #472: Allowing add Laser Intensity variable in TOOL ON GCode

Tibus, 2018-04-24:
Hello, 
To be compatible with my CrealityCR10 (Marlin) and modded laser, I just needed a way to turn on and off tool in GCode with the M106 Command.
You can define it in TOOL ON/TOOL OFF textfield but there is no way to get back the laser intensity dynamically. 
I've so just cloned the code and add a way for it:
In TOOL ON, you can now add the $INTENSITY variable that will be replaced during gCode generation by the "LASER INTENSITY" field and its value.
Like : 
`M106 $INTENSITY`
will become with the intensity to 80% and Laser intensity to "S"
`M106 S80`

It's really simple but do the job amazingly on Marlin ;D

Are you interested about a pull request or maybe another way to do it simpler?

jorgerobles, 2018-04-24:
@cprezzi was this kind of addition planned for the server?

jorgerobles, 2018-04-25:
@Tibus I've been checking the server and has no deal with kimbra in gcode execution only connection, so your pull request is welcome! Go on :)

One thing to take account (without having seen your PR) is that gcodeToolOn is used in several places, at **gcode generation** and also at **jog**. if not done I propose to create a new module (lib/postprocessor?) with decorator functions (now just one, a replace alias) to be used to wrap all gcodeToolOn appearances.

cprezzi, 2018-04-26:
Sure it would be possible to do a "post processing" (code replacement) on server side to make gcode "marlin" compatible, but i think we a more flexible if we do that on gcode generation in the frontend.

jorgerobles, 2018-05-01:
Closing due gcode refactoring by @Tibus .. Thanks!

closed 2018-05-01 by jorgerobles