LaserWeb/LaserWeb4 #483: Spindle Gcode incorrect format. Laser Gcode works!

carlosen14, 2018-05-05:
I own a CNC Multifunction machine, laser on and spindle on pins are connected separately and each works with different Gcode command, laser on M3 SXXX, laser off M5, spindle on M3 SXXX and spindle off M5. This machine works with Marlin-FW and i did enable the "Intensity separate line" checkbox to get it work.

When generating gcode to work with laser, all works just great.. Did a few test and thats all..
Moves, Feedrate, Tool On, Tool Off, etc..

But when I want to generate gcode to use the spindle, it doesn't appear M3 SXXX to turn on the spindle, just appears SXXX and also it appears in the same line.

Spindle GCode Failing
```
G21         ; Set units to mm
G90         ; Absolute positioning
G28 X0 Y0 ; Home X and Y axis
M5       ;  Enable Laser/Spindle (0 power)

;
; Operation:    0
; Type:         Mill Pocket
; Paths:        2
; Direction:    Conventional
; Rapid Z:      10
; Start Z:      0
; End Z:        -3
; Pass Depth:   0.5
; Plunge rate:  500 mm/min
; Cut rate:     80 mm/min
;
; Retract
G0 Z10.000

; Path 0
; Rapid to initial position
G0 X5.152 Y3.159
G0 Z0.000
; plunge
G1 Z-0.500 F500 S255 ; HERE!! It is supposed to put M3 S255 on the previous line..
; cut
G1 X5.151 Y3.159 F80 S255
G1 X5.151 Y3.160
G1 X5.150 Y3.160
G1 X5.148 Y3.161
```

Laser GCode working correctly, even works fine the PWM values
```
G21         ; Set units to mm
G90         ; Absolute positioning
G28 X0 Y0 ; Home X and Y axis
M107       ;  Enable Laser/Spindle (0 power)

; Pass 0

; First Move
G0 X25.75 Y0.65
M106 
; Generated by LaserWeb (lw.raster-to-gcode.js)
; Size       : 0 x 0 mm
; PPI        : x: 300 - y: 300
; PPM        : x: 0.0846666667 - y: 0.0846666667
; Tool diam. : 0.1 mm
; Feed rate  : 500 mm/min
; Beam range : 0 to 244
; Beam power : 20 to 100 %
; Options    : trimLine, joinPixel

G1 F500

M106 S0.0000
G0 X25.75 Y0.65
M106 S50.3310
G1 X25.80
M106 S52.6275
G1 X25.90
M106 S54.1584
G1 X26.00
M106 S53.6481
G1 X26.10
```

[workspace.json.txt](https://github.com/LaserWeb/LaserWeb4/files/1976994/workspace.json.txt)

[laserweb-profiles.json.txt](https://github.com/LaserWeb/LaserWeb4/files/1977006/laserweb-profiles.json.txt)

jorgerobles, 2018-05-06:
Marlin FW is partially implemented. @tibus?

carlosen14, 2018-05-07:
what can i do in this case?.. i mean, is it possible to create a Machine Profile to use the Spindle and put on the **Gcode start** to turn on the Spindle and turn off on the **Gcode end**. And change the Machine Profile to another one to user the Laser, where i will put the correct values in **Tool on** and **Tool off** also **Laser intensity** for PWM use..

This described is what i am actually doing but i have to modify it every time depending what i want to use.. But can i control it just by changing the machine profile?

cprezzi, 2018-05-08:
@carlosen14 Yes, that's exactly how it works. Just prepare a machine profile for every configuration you need so you can easily switch between them.

carlosen14, 2018-05-08:
Ok, but i am actually not able to do so, i tried it already. I'm on Mac OS 10.13.4 using Laser Web 4.0.991.

I did create my two profiles Laser and Spindle, but i don't make it work the part of changing machine profiles. I really don't know the process or the function of **Apply** and **Update** buttons. 

[image]

When I select another profile and press **Apply** it changes absolutely nothing. And if press **Update**, in details, it changes the profile name an mixes the profiles. So, I ended up messing the profiles because it doesn't save the Gcode commands on each box, it's like if only mess the profile names.

I did watched some YouTube videos to get into but i am still not able. So, I'll continue searching, thanks for the info.

By the way, great tool!

cprezzi, 2018-05-09:
- Start by selecting a generic machine profile in the dropdown
- Hit apply
- Click ok in the popup dialog
- Make your changes in the settings
- Enter a new profile name in the "New Profile" field
- Click "+" button

You should now see the new profile in the profile dropdown and be able to reactivate at any time.

closed 2018-06-23 by jorgerobles