LaserWeb/LaserWeb4 #484: Proposal: recognise <title> tag in .svgs

lucalcin, 2018-05-21:
Hi,

.svg standard allow the use of the tag <title> to name each element.

When working with large .svg files, can be useful to show in the document list a name for each path or element.
Just as an example, I'm working now with a .svg that contains more than 2500 elements, grouped in more than 100 objects, each with his own name. Some elements are to engrave with 20% power, others with 50%, other to cut. Without names, is tricky to understand what element is to engrave and what is to cut.

In attachment, a zipped .svg file with named elements and a concept image showing the names in the document section.

Thanks

Luca

[image]

[title example.svg.zip](https://github.com/LaserWeb/LaserWeb4/files/2023995/title.example.svg.zip)

jorgerobles, 2018-05-22:
Thanks, will try to.
I do some complex cuts too but solve with color filtering. Prepare your file asigning stroke colors by operation and ordering, and then add the whole file as an operation (several times) using the stroke color filter. Works like a charm.

lucalcin, 2018-05-24:
REALLY POWERFUL! The color filter is a really great feature. I already used color to identify type in my job (purple for cut, green for engrave and so on). This simplifies complex operations.
Anyway, naming convention will be useful when in a single drawing lies multiple objects (i.e. ribs of a wing or of a boat), so you can identify and choose the one to work.

jorgerobles, 2018-05-24:
it's done already, will be available on the next build