LaserWeb/LaserWeb4 #489: Laserweb 4.0.993 locking up

marshalltjones, 2018-06-07:
I am using Laserweb 4.0.993 on Win 7 running on an Intel i3 with 16 GB of ram.  I loaded a 300 dpi png (292.6 mm tall 195 mm wide) in Laserweb and set it to 200 mm tall.  0.2 mm laser width.  When I generate g-code, the Laserweb window goes completely white, and you have to restart the program.  When I drop the file to 100 mm tall, it generates g-code without an issue.  

I am including the file and a screenshot of the settings before g-code generation.  

Am I missing something?
[image]

marshalltjones, 2018-06-07:
[image]

jorgerobles, 2018-06-07:
[image]

Sorry, I guess is a processor performance issue. Your pic goes fine (mac, i7, 16Gb Ram)

marshalltjones, 2018-06-07:
I'll have to try it on my i5

closed 2018-06-23 by jorgerobles