LaserWeb/LaserWeb4 #490: windows crash - blue screen of death

nicolasdb, 2018-06-15:
Hi guys, 

I don't know if it's hardware or software related but I had the problem on 3 differents computers, on windows 8.1 and 10. 
With the last build and the one before. I did tried to upgrade but had the problem again. 

It usually work fine, I can engrave, generate gcode, with different svg but I had the crash at least in two scenario. 
- just at the end of the "generate" operation. It finish to calculate then crash windows. 
- and at the end of some jobs. Laserweb stop the job then crash windows.

The machine is a small cheap chinese one, 2.5W diode laser like Eleksmaker. I flashed the grbl 1.1 on it. 

Any idea where than could come from? 

I'm wondering about the machine though .... that's the only common link between event.

thanks for advice.

jorgerobles, 2018-06-15:
The issue comes from CH341 (the usb chipset if your board) drivers for
windows.

Has been answered several times on g+ LW community.

 it's not related with LW, any other software will rise the BSOD.
Please contact your machine provider to solve the issue.

Sorry for the inconvenience.

nicolasdb, 2018-06-15:
ok cool, I didn't look on the G+ sorry. 

It's exactly this driver. I'll look there then; 

Thanks for your help.

closed 2018-06-15 by nicolasdb

tsipa, 2019-03-06:
Same here. I can successfully run candle connected to comport for days and it'll work, but running laserweb for few hours crashing my PC within hour or two.

cprezzi, 2019-03-07:
@tsipa Don't expect to get an answer on a already closed issue!

jovica94, 2019-04-15:
Since there's no G+ community anymore for LW I would like to know where can I find the solution for this problem(BSOD). @nicolasdb Have you fixed it? Sorry for my bad English...

nicolasdb, 2019-04-15:
well yes and no, @jovica94  
I didn't find any solution for the driver on Windows but I'm now using an old laptop on linux for machine control and I didn't got anymore problem since.

tsipa, 2019-04-15:
@jovica94 there is no solution for the driver cause driver works more or less fine, use this https://github.com/LaserWeb/lw.comm-server/wiki/Manual-installation-(Windows) manual to install latest version of LW which have proper version of node-serialport onboard and will work

tsipa, 2019-04-15:
here is some more details https://forum.makerforums.info/t/bsod-windows-10/54492

darkracer125, 2019-04-20:
> The issue comes from CH341 (the usb chipset if your board) drivers for windows. Has been answered several times on g+ LW community. it's not related with LW, any other software will rise the BSOD. [...]

i'm sorry but that's a load of crap.  
it's 100% laserweb that is causing it.  
none of us get bsod with anything else.  and we get bsod on multiple platforms.  
laserweb IS the cause of the bsod.  
so it's simple.    

attention everyone.   stop using laserweb.  the software is causing problems and the developer is denying it.  
stop using laserweb.   even inkscape is better.

darkracer125, 2019-04-20:
> @jovica94 there is no solution for the driver cause driver works more or less fine, use this https://github.com/LaserWeb/lw.comm-server/wiki/Manual-installation-(Windows) manual to install latest version of LW which have proper version of node-serialport onboard and will work

what is this bullshit about starting a server and a webfront?   
nobody uses a server.  

people are running the program.  they are pressing print. and anywhere between 5 minutes and 30 minutes later the computer gets a bsod.

tsipa, 2019-04-20:
You misunderstood what i'm saying and claiming it's a BS, what a BM. By starting Server you can use the most recent version of LW from source, which is not available in binaries for now. I'm nearly confident you won't be able to build LW binaries unless you're expert in MS Visual studion and surrounding ecosystem.

BTW, the problem is caused by underlying lib node-serial, the fix is merged https://github.com/LaserWeb/lw.comm-server/commit/bdee9271dd83ccacb00d8b9a121d449ff1c3fdae it's just not available as binary for windows.

cprezzi, 2019-04-23:
> @darkracer125 wrote: 
> - i'm sorry but that's a load of crap.
> - what is this bullshit about starting a server and a webfront?
> - nobody uses a server.

**Calm down or I will block you!!**
People like @tsipa are just trying to help you and you should estimate that!

What you wrote is showing that you don't have any clue what you are talking about. Don't you understand the difference between quickly streaming a lot of data over USBserial and opening a terminal conection for slowly typing some commands? 

The problem you describe is very rare and only happening on some PC and controller board combinations. It's mostly happening with bad quality (long) USB cables and low quality CH340 USB chips on the controller board. It can be solved by a newer version of the Serialport library that LaserWeb is using (but not developling). I have already changed the LW backend server (lw.comm-server) to use the newer Serialport library v.6.2.2, but this has not yet been included in the binary installer. If you want to use the patched version, you need to install the backend server software manually (can be installed on the same pc).

jovica94, 2019-04-24:
I have an update regarding my problem with BSOD - haven't tried using it with RPi and server or with new hardware(Arduino Nano) yet... All I did is:  - [http://www.wch.cn/download/CH341SER_ZIP.html](url)
                                                                           - Device Manager/COM3/Set Baud rate to 115200
It seems BSOD is gone for now... But I like the idea of using terminal connection for machine control, so that's what I going to try next.