LaserWeb/LaserWeb4 #492: Wont add end tool to gcode

garybelcher, 2018-06-18:
Having problems trying use raster on my converted 3d printer (cut works fine)

its a mightyboard (ctc)

I just cant get it to add the gcode tool off command m127

hopefully added everything needed (wont let me upload the .svg so uploaded the png instead 

( little add on raster works fine from the plugin with inkscape) 

[workspace.json.txt](https://github.com/LaserWeb/LaserWeb4/files/2111478/workspace.json.txt)
[image]
[image]

garybelcher, 2018-06-20:
Do I need to bump this? 

Hoping for a fix.

jorgerobles, 2018-06-22:
> Anyone???

Man, don't want to be unpolite, but first of all, maintainers, namely me,
don' t do this for a living, so please be patient. Also, LW is open source
and free. Fork an contribute, and will grow better.

Second, the issue you opened is, although arising to 3d printers becoming
cnc not aplicable in all cases. Proper firmwares already handle that nicely.

Third, I have already made a patch two days ago, but have to check if is
coherent with the rest of the application to publish the binary, you know,
there are a bunch of other people using it too.

I expect to check the code over the weekend.

Regards

garybelcher, 2018-06-22:
Apologies didn't mean to come across inpatient. I'll be happy to contribute when it works for me. I guess if I didn't post up and state the problem you maybe wouldn't of known about it.(saying that have since seen others have same problem as me)

I don't understand all the technical bits and bobs but do appreciate all the work that has gone into what looks like to be amazing software.

Will there be an update. How will we no about the fix? If it's fixed?

jorgerobles, 2018-06-23:
Fixed
https://github.com/LaserWeb/LaserWeb4-Binaries/releases/tag/v4.0.994-130

closed 2018-06-23 by jorgerobles