LaserWeb/LaserWeb4 #494: FEATURE: Load/Save settings from/to server

quintesse, 2018-06-22:
Right now setting up several clients/browsers is quite a hassle. Yes you can save all the settings as files but then you still have to get them to the other clients.

Right now I have to:

 1. Save the files
 2. Locate them
 3. Mail them
 4. Save them locally
 5. Load them

And you'd have to do that each time you make any changes (or remember to make the exact same changes everywhere).

Being able to save the settings on the server itself somehow would be very useful I think.

PS: Personally I'd be more than happy if this was a just a single "save slot", I myself don't have any use for multiple sets of machine profiles and settings on a single server, but perhaps others do?

jorgerobles, 2018-06-23:
Hmm could be done thru https://www.npmjs.com/package/google-drive but it's not work of an evening or two :P

quintesse, 2018-06-23:
Isn't there a way it could be done on the server itself?

quintesse, 2018-06-23:
I opened https://github.com/LaserWeb/lw.comm-server/issues/62 and updated the description of this issue somewhat.