LaserWeb/LaserWeb4 #495: Support for machine "benbox laser 3.1(c) mul 2015.06.06" ?

df8oe, 2018-06-25:
Hi,

actually I am fighting with the second step. LW4 is properly installed and is starting up at Debian 64 bit.

I can connect to my laser cutter via USB and can see in the log window of LW4:

> Machine connected
> benbox laser 3.1(c) mul 2015.06.06
> unknow M115
> unknow M115
> No supported firmware detected. Closing port /dev/ttyUSB1
> Machine disconnected

It identifies itself via USB with "vendorId":"0x1a86","productId":"0x7523". It seems to be a machine which is very often sold from Chinese sellers - it is like this one:
https://www.ebay.de/itm/Mini-Laser-Engraving-Machine-30X40CM-DIY-Logo-Cutting-500mW-Marking-Wood-Printer/282830912097

It has Arduino nano as controller. It is sold with "Benbox Windows Software" which does not work under my Linux system (I got it partially working using wine bur do not want to use this soft). Is there any chance to get it working?

jorgerobles, 2018-06-25:
It's surely a grbl, if not, check youtube for grbl for benbox

df8oe, 2018-06-25:
Yes, I supposed that already and tried with setting "grbl" - sorry, I forgot. Controller seems to be a modified EleksMaker® ManaSE 2 controller. Laser PWM connection is not applied - laser and fan are fed through 12V 2pin connector. Bur Laser can be regulated by software so the 12V connector isn't a "straight one". Before flashing new firmware (and reverse engineer what is about laser power management) I am looking for a solution to get it run without fw changing. What means "unknown M115" in the logs? Is is coming from controller to LW4 or other direction?

jorgerobles, 2018-06-25:
Yes, it comes from the controller, m115 is a gcode sent by the dectection
of the firm (guessing is marlin)

df8oe, 2018-06-25:
Thanks. Meanwhile I stuck up and swapped the Arduino nano to one containing grbl and all works fine. The stuff which was on original board is weird ;)

closed 2018-06-25 by jorgerobles

Haard-Shah, 2019-02-10:
@df8oe did you ever get it to work with the original board that comes with the kit? I am having the same issue and would like to run laserweb for my laser cutter, the Benbox software is very sketchy and not well thought out.

df8oe, 2019-02-10:
@Haard-Shah : as I already have written in my last post I have replaced the firmware. I never ever have tried something more with th original firmware - it is weird and ugly...

electrizzato, 2020-01-11:
> @Haard-Shah : as I already have written in my last post I have replaced the firmware. I never ever have tried something more with th original firmware - it is weird and ugly...

hello,
I'm having your same problem (Machine connected benbox laser 3.1(c) mul 2015.06.06 unknow  115 unknow M115) , could you help me step by step in replacing the firmware? 
thank you

mhguidry, 2022-09-14:
> Thanks. Meanwhile I stuck up and swapped the Arduino nano to one containing grbl and all works fine. The stuff which was on original board is weird ;)

Df80e: could you send link of replacement NANO that got you running again. Also, any more information on general running of laser since you got it going. TIA.