LaserWeb/LaserWeb4 #497: BUG: action2gcode: Invalid Regex flags

riaancillie, 2018-06-28:
With a raster image select and a Laster Raster operation being created, gcode generation will fail if the machine configuration (Settings > GCODE) has "Tool On" and "Tool Off" gcode defined. 

```
abstract-generator.js:11
Uncaught SyntaxError: Invalid flags supplied to RegExp constructor 'gis'
    at RegExp (<anonymous>)
    at DefaultGenerator.postProcessRaster (abstract-generator.js:11)
    at postProcessing (cam-gcode-raster.js:123)
    at Worker.r2g.onmessage (cam-gcode-raster.js:238)
```

In commit It appears that "gis" flags are invalid, specifically the single-line flag which is not supported in javascript.

[LaserWeb4/src/lib/action2gcode/generators/abstract-generator.js #492](https://github.com/LaserWeb/LaserWeb4/blob/d3583016cd9fa7f5d478471769cfdf71065a8db8/src/lib/action2gcode/generators/abstract-generator.js)

[Laserweb-Workspace.json.txt](https://github.com/LaserWeb/LaserWeb4/files/2144776/Laserweb-Workspace.json.txt)
[image]

jorgerobles, 2018-06-28:
Download previous version. Will drop the release.

jorgerobles, 2018-06-28:
Anyway is weird I had successfully run that piece of code on latest chrome.
Have to check chromium version

jorgerobles, 2018-06-28:
https://github.com/LaserWeb/LaserWeb4/commit/289dec3a90467d38c461841d88c2a9e54a64b7d3

closed 2018-06-28 by jorgerobles