LaserWeb/LaserWeb4 #499: Marlin issue with raster

SmarteedCS, 2018-07-02:
Hi,
I'm supper happy with the new Marlin support, thanks a lot.
Laser fill path and laser cut is OK, but I have some trouble with the raster gcode generation.
Here is a sample of laser cut outside gcode, M106 P0 SXX is OK:

> ; Pass 0 Path 0
> G0 X54.03 Y-0.10
> M106 P0 S253.00
> G1 X54.04 Y-0.10 F500

And here is the sample of a gcode generated with raster:

> ; stripped: S0.0000
> G1 X227.70 Y0.70
> ; stripped: S1.9843
> G1 X227.80
> ; stripped: S8.9294
> G1 X228.20
> ; stripped: S0.9922
> G1 X228.30
> ; stripped: S0.0000
> G1 X228.50 Y0.90
> ; stripped: S18.8510

Anyone had the same problem ?

[LW_WP_raster_issue.txt](https://github.com/LaserWeb/LaserWeb4/files/2154965/LW_WP_raster_issue.txt)

cprezzi, 2018-07-02:
The raster module does not support marlin specific code generation and we don't have a dev yet who wants to work on that module. Sorry!

SmarteedCS, 2018-07-04:
For people interested in making it work by hand, here is what I do.

Laser Raster configuration:
Trim Pixels enabled
Join Pixels enabled
Burn white disabled
Verbose GCode disabled

1.Genrated Gcode, save it, open it with for example notepad++.

2.Select "; stripped:" ang replace all by your laser enable command ( for me "M106 P0" )

3.If you want to move faster over blank area:
  delete "G1 FXXX" line
  replace all "G1" by "G1 FXXX" (XXX is your engraving speed)
  replace all "G0" by "G0 FXXX" (XXXX is your mooving speed)

4. Save file and reload it in LW

schme16, 2018-10-02:
Is this issue still in effect? When I output raster for marlin I now get M106 and G1 commands

SmarteedCS, 2018-10-02:
Yes, for me, with LWV4.0.996 and Marlin Gcode generator the issue is still in effet.

christopherwit, 2018-12-09:
How do you guys even get gcode? In my case using marlin the progress bar shows 100% before reaching the end of the progress bar. At the same time Laserweb gets stuck and doesn't respond to anything.

cprezzi, 2018-12-25:
The latest LW is capable of generating the laser power commands as separate lines also for raster engraving, so it's compatible with marlin!
You just have to go to settings/gcode and set "TOOL ON" to "M116 P0", "TOOL OFF" to "M116 P0 S0" and activate "INTENSITY SEPARATE LINE".

closed 2019-06-14 by cprezzi