LaserWeb/LaserWeb4 #500: Duet Support

keyz182, 2018-07-09:
What would need to be done to add support for the [Duet ](https://duet3d.com/) to LaserWeb? I'm willing to have a go, but not quite sure where to start.

cprezzi, 2018-07-10:
There are two requirements. 

The first is, that the lw.comm-server can detect the duet firmware. Does the duet firmware send a firmware recognition string on connect or can it be queried with a serial command?

The second is the way how the laser power is set with gcode. Laser capable firmwares should accept S values as params of G1-G4 commands. That is ideal for engraving. Other firmwares like marlin do need a separate gcode line for the laser power (like M4 S128), which need some time and therefore slows down engravin a lot. Which type of gcode does the Duet firmware accept?

keyz182, 2018-07-10:
Following up, it appears Duet uses the S parameter on G1 for endstop checks, isn't present on G2/G3, and is used for time to wait in seconds in G4.

What does the S parameter do on other firmwares? Is it just setting the intensity (e.g. S0-255, or S0-1) or a simpler on off?

I could look at adding support for a similar parameter to the firmware (though it would have to be a different letter than S) if e.g. the "activate laser on E increasing" isn't good enough.

keyz182, 2018-07-11:
[It appears support for S parameter on G1 is coming in 2.0.2 of reprapfirmware](https://forum.duet3d.com/topic/5821/major-performance-problem/12), so that's that bit sorted. I'll take a look at what's needed in lw.comm-server.

T3P3, 2018-08-11:
@keyz182, thanks for starting this conversation! I am one of the Duet creators, @FabCreator put me onto this topic. I am not familiar with Laserweb but its obvious a tool that a lot of people use for lasercutting/CNC and it would be very cool if it supported the Duet. For faster and more reliable operation it would be good if Laserweb supported a "send to Duet" option so the whole file could be stored and read from the on board SD card on the Duet.

@cprezzi feel free to drop me a line info@duet3d.com if having a Duet would help you with supporting it better.

cprezzi, 2018-08-13:
LaserWeb does not yet support sending the file to the sd card of the controller, because we always wanted to be a "realtime" frontend and not another gcode sender. This is especially important on cnc milling and long vector lasercuts. In my opinion, the frontend should be able to pause/resume/stop the job immediately (no waiting for commands in the queue) and do feed and laser power changes in realtime, like we can do with Grbl.

Of course I'd like to support the Duet as one of the more advanced controllers. There are "just" two main requirements:

1. We need a way to detect the firmware. The M115 query/response seems ok for that.
2. For fast raser engraving, the PWM value should be in the G1-G3 command (like G1 X100 Y0 S100).

If the gcode "dialect" is close to grbl or smoothieware, it could be quick to implement. Having a Duet for testing would clearly speed up the process ;)

cprezzi, 2018-08-13:
That sounds very promising. I always sugested to add a "line of pixels" command to reduce overhead for raster engraving.

cprezzi, 2018-08-13:
Do you also adjust the laser pwm depending on x/y acceleration and deceleretion (like Grbl in M4 mode)?

FabCreator, 2018-08-13:
@cprezzi drop me a mail at bonne@fabcreator.com and I will organise you a board. :)

T3P3, 2018-08-14:
I checked and apparently the reduction in power based on acceleration is not yet implemented in this Beta but will be before the beta makes it into an actual release.

cprezzi, 2018-08-15:
@FabCreator Thank you for the offer. Tony already told he will organize a v1.03 board for me :)

FabCreator, 2018-08-16:
@cprezzi is there a way to send realtime commands even if the file is running direct? For long files USB is just not reliable enough so creating a send to would be a great addition to LW. I thought the main reason this was not explored futhure was because of smoothie unable to read and write reliably in this regard. (This was back in LW3 with peter)

cprezzi, 2018-08-16:
@FabCreator It depends on the firmware, if "realstime" commands are accepted during SD job execution. I think RepRapFirmware has some realtime commands (like pause/resume) that can be used while a job is running. @T3P3: Is that correct (also via USB)?

We was also talking about connecting the Duet via ethernet (LAN/WLAN), exactly like the DuetWebControl is talking to the firmware.

I think it would not be enough to just add a "send2SD" function. What happens when the filename already exists? I think we would need to implement the whole SD support (dir, cd, store file, read file, delete file).

T3P3, 2018-08-16:
@cprezzi yes some commands are not buffered to allow for faster pause/resume.

If you are accessing the Duet over HTTP then the commands to do the file handling are already in the API.

Wingeddonkey, 2018-09-07:
Looking forward to watching and seeing the progress on this. 
Recently upgraded to a duet and am hopeful to see the support added very soon. (So I don’t have to leave two boards installed for laser and printing)

cprezzi, 2018-09-07:
I have already finished the implementation of USB serial communication for Duet in lw.comm-server and merged it to the master branch. The SD functionality is also implemented into the server and the API was extended, only the frontend part is still missing.

keyz182, 2018-10-07:
@cprezzi what's required for the frontend?

keyz182, 2018-10-07:
@cprezzi what's required for the frontend? I can have a look and try to implement what's needed, but may need some guidance.

cprezzi, 2018-10-08:
@keyz182 The frontend needs an option to manage the SD card content (list folder, cd, upload or delete files).

cprezzi, 2018-10-08:
No, the USB connecttion is handled by the NodeJS server (lw.comm-server) that is compiled into the exe. The client (frontend) has to use the server API on the websocket connection (as documented in https://github.com/LaserWeb/lw.comm-server/wiki/API-(for-client-communication).

You can find the current frontend for machine control in https://github.com/LaserWeb/LaserWeb4/blob/dev-es6/src/components/com.js. What would be nice is to add a "manage SD" button and popup dialog with a file explorer to navigate throug the sd card, be able to create/delete folders and upload/delete files.

cprezzi, 2019-01-15:
@Wingeddonkey I think you should better ask on the RepRapFirmware forum for differencies in gcode details. 

As far as I understand, the gcode interpretation in RepRapFirmware should be very close to Smoothieware, so saving the gcode manually and uploading to the SD card for execution should work.

The newest binary is always published on https://github.com/LaserWeb/LaserWeb4-Binaries/releases.

closed 2019-06-14 by cprezzi