LaserWeb/LaserWeb4 #501: Support Power-ON delay for Laser Cutter/Raster operations

omuzychko, 2018-07-10:
I am  trying to use **LaserWeb4** for the first time and I think it has a great potential. 
At the moment my main software to generate gcode is **Inkscape** with an extension from **J Tech Photonics**. This setup is lacking on ability to **Laser Cut Inside** what is so nicely implemented in **LaserWeb4**. It is extremely critical feature for PCB DIY (cutting SMD stencils from Mylar or photo-resist exposure: [Laserweb-Workspace.json.txt](https://github.com/LaserWeb/LaserWeb4/files/2182199/Laserweb-Workspace.json.txt)).  But after trying **LaserWeb** I've found that and I am missing one major feature supported in **Inkscape/JTech** setup - it is a possibility to introduce a variable delay after we turned the laser ON (`M106 Snnn`) and before we start moving it (`G1 ....`):

[image]

This is achieved by **dwell** command (`G4 Pnnn`). It ensures that all the previous commands in the queue have been accomplished and introduces optional delay before moving on to the next one. Here is the gcode output from **Inkscape/JTech**:

```gcode
...
; Move the Laser to the starting point of the cut
G1 F3000                
G1  X29.09 Y17.83
; Make sure that the movements are done before we turn the Laser ON
G4 P0       
; Turn On the Laser at 100% power               
M106 S255      
; Delay for 50 ms, so that we ensure that the laser fully turned on and we initiated a cut
G4 P50  
; Start moving the laser to make a cut         
G1 F750.000000     
G1  X29.09 Y16.72
; Make sure that the processing movement-commands queue has completed
G4 P0           
; Turn Off the Laser
M107 S0     
;  Move to the starting point of the next cut            
G1 F3000                
G1  X28.24 Y19.55
...
```

Why this delay is needed:

- Cheep laser modules from China have poor electronics quality and might require some transition time before they start outputting requested power;
- When trying to cut materials with low absorption levels - sometimes a major delay needed at origin point of the cut to initiate the burning process. Say for clear or light-colored plastic when you lite-up the laser - it would not start burning through immediately. It will take some fraction of a second to absorb enough energy to burn down the hole. But when done - you get some dark carbon from burned plastic on the side-walls of this hole. Because of this dark carbon the efficiency of energy absorption jumps up severely, so now you may start moving fast-enough to make a clean narrow cut and not over-burn on the sides of the cutting trajectory.

In other words - without this feature the clean cutting of clear or lightly-colored materials with cheap 2..5W laser modules from eBay is virtually impossible. It either does not initiate a cut or over-burns the cutting path.

The value of the `G4 Pnnn` delay is closely correlated with chosen laser power (`M106 Snnn`) and cut rate (`G1 Fnnn`). So it should be property for each specific Laser Operation, textbox input next to the "Cut Rate"  (NOT in a global GCode generation Settings):

[image]

cprezzi, 2018-07-11:
You can define the "TOOL ON" and "TOOL OFF" commands in settings/gcode (multiple lines are valid).
That should solve your problem.

omuzychko, 2018-08-03:
@cprezzi - your suggestion works and this is the first thing I've tried. And it is only OK if the delay doesn't depend on actual power of the laser applied. But it does. For 5W laser I need to introduce up to 100 ms dealy when cutting at full power, and somewhat 30..40 ms delay when running at 10%. If I use the same laser power for all operations - your solution works OK. But when each operation should have different laser power (some to cut through, others - to raster some shading at lower power) - the delay  should be different as well.

cprezzi, 2018-08-03:
Vielen Dank für Ihre Nachricht.

Die COMsulting GmbH macht vom 23.7. bis 7.8.2018 Betriebsferien.
Wir bitten um Ihr Verständnis, dass in diesem Zeitraum weder Anrufe noch Emails beantwortet werden können!

Ab dem 8.8.2018 sind wir dafür wieder voller Elan für Sie da!

Schöne Grüsse
Claudio Prezzi

COMsulting GmbH

gabrielgrant, 2021-07-16:
Running into this same issue. I've added a fixed delay with a `G4` "dwell" command in the "TOOL ON" field, wich kind of works, but this seems to apply at the beginning of every line segment (even when the cut continues from the same point as the previous segment, like in a corner). the result is extra time spent (and kerf width) at each corner