LaserWeb/LaserWeb4 #507: "RangeError: Invalid string length"

professorofmit, 2018-08-01:
### Follow this steps to open an issue ###

- [ X] If refers software, include `Workspace.json`.  This is available on *CAM tab top, Workspace -> Save*. Prepare your work, save `workspace.json`, rename as `workspace.json.txt` and attach in this issue. *All resources regarding the issue should be attached (SVG files, images, etc.). An issue opened without resources will be automatically closed upon inspection*

- [ ] If refers hardware, include enough images, log files, and optionally videos (or link to a Cloud Drive folder) to inspect.

I got the following error when uploading the included SVG files into Laserweb, and I wanted to learn how to solve this issue.

The file has serious issues. If you think is not your fault, report to LW dev team attaching the file.
"RangeError: Invalid string length"

[laserweb issue.zip](https://github.com/LaserWeb/LaserWeb4/files/2248036/laserweb.issue.zip)

professorofmit, 2018-08-01:
I tried opening my SVG files at https://laserweb.github.io/LaserWeb4/, and I couldn't get them to work there either. Chrome displayed the following error: "Aw Snap! Something went wrong while displaying this webpage."

lischenko, 2020-06-05:
In case it helps with debugging, I am seeing `"RangeError: Invalid string length"` with this 60KB svg file in LaserWeb 4.0.996 on a Mac.

[Untitled-F_Cu.svg.zip](https://github.com/LaserWeb/LaserWeb4/files/4738957/Untitled-F_Cu.svg.zip)

lischenko, 2020-06-06:
I tried minimizing the svg contents that still trigger the issue for me. It does not seem to be directly caused by the file size.
For example, here is a small 507 byte long svg file that still triggers the `"RangeError: Invalid string length"` message for me. It has one big circle (and a few empty groups) so I theorize that the issue may be caused by the length of its interpretation (many tiny arcs or something). Someone who knows anything about LW might see more into it.

[small.svg.zip](https://github.com/LaserWeb/LaserWeb4/files/4739182/small.svg.zip)

cprezzi, 2020-06-08:
That small.svg also doesn't work when opened in Inkscape! How did you create it?

cprezzi, 2020-06-08:
@lischenko I have analyzed the Untitled-F_Cu.svg and the problem is, that the objects are not closed pathes but single lines. For example, all the middle pads are just lines with 2mm width. That could not work for vector engraving as the laser has a fixed with of 0.1-0.3mm. For vector engraving, your file must only have thin contours (like 0.1mm) and no fill color. An infill must be made with a "Laser Fill Path" operation.

For PCBs, the easyer solution is to save the file as a JPG or PNG and use the "Laser Raster" operation.

lischenko, 2020-06-09:
Thank you for looking into it, @cprezzi !

The small.svg does open in Inkscape v1.0b2 on Mac OS for me (needs a lot of zoom out else the entire view is black - there is a giant circle after all). 

I created small.svg by progressively removing and modifying XML nodes of the Untitled-F_Cu.svg. Just in case, here are some intermediate, larger files in case they are useful - they all fail to open in LW for me: [svgs.zip](https://github.com/LaserWeb/LaserWeb4/files/4747544/svgs.zip)

Also, the small.svg does open in LW if I remove any of the empty groups or reduce the radius of the circle.

The explanation for Untitled-F_Cu.svg explains a lot to me! Still, I would expect the same message as for empty files today ("Error: No supported tags found."). 
For context, Untitled-F_Cu.svg is an example of what KiCAD (Pcbnew) outputs from its Plot operation, too bad it is not useful out of the box. I'll find a way to turn thick lines into countours, but that's an extra prep step.

The raster workaround works, but it has disadvantages: slower to engrave and the lines are not as sharp.

To summarize, I can see some short term potential improvements for LW:
- Produce more detailed error messages for inputs such as small.svg and Untitled-F_Cu.svg
- [not terribly important] Provide some indication of analysis (it takes ~10 seconds to produce an error message for small.svg, and all that time the UI is silently unresponsive)
- [probably long term] Ideally, as a user, I would like those thick lines to be turned into contours automatically by LW.