LaserWeb/LaserWeb4 #509: Firmware not loading more than once.

fergusfp, 2018-08-18:
I am having an Issue with laserweb4.0 and my MKS BASE V1.3 on mac OSX.

I follow the instructions on the website and put the current copy of the firmware for grbl-LPC v1.1 on to a micro-sd card and rename it. I eject it from the computer and place it in the MKS BASE and plug it into my computer. After it boots I can connect it through the laserweb interface and the terminal tells me that is connected and it has found grbl 1.1 and both the $ and $$ commands work. 

[image]

However as soon as I reset, power cycle, or unplug the board and try and reconnect it tells me I don't have the correct firmware available. 

[image]

I have tried several different cards from a 32GB scandisk to a 128Mb no-name and all have the same results. The only way to get a connection back is to place a fresh version on the card and remove the FIRMWARE.CUR

I have also tried the previous version of grbl-PLC

Any help would be much appreciated.

fergusfp, 2018-08-19:
This Morning I found a link to this conversation https://github.com/LaserWeb/LaserWeb4/issues/322 and so I altered my .env file to include 

RESET_ON_CONNECT=1
GRBL_WAIT_TIME=10
FIRMWARE_WAIT_TIME=30

But still no difference for me. 

Can you please tell me if I am putting the .env file in the place? I have included this photo, 
[image]
Thank you

fergusfp, 2018-08-19:
Just to confirm the board and firmware are working correctly, outside of laserweb coolTerm has no problem talking to the board. 

[image]

But still no communication with laserWeb after initial boot.

Thanks

cprezzi, 2018-08-20:
Grbl-LPC (like every grbl) sends a welcome string after each USB connect. LaserWeb is waiting for this string to identify grbl (for communication compatibility) and if the string is not received within GRBL_WAIT_TIME the connection is closed.

Do you use a direct (good quality) USB cable between the Mac and the MKS board? Or some USB2serial modem?

cprezzi, 2018-08-28:
I also use a MKS SBASE V1.3 with grbl-LPC, but with Raspberry Pi as Server and I don't have these problems.

cprezzi, 2018-08-28:
I remember that there was a problem with OSX and the .env file. 

@jorgerobles do you remember if this problem was solved an how?

closed 2018-12-11 by cprezzi