LaserWeb/LaserWeb4 #510: SyntaxError: Unexpected token Raspberry pi install

Benjin454, 2018-08-26:
Info:
- Raspberry pi 3 model B
- Running raspbian stretch
- Following install instructions from https://github.com/LaserWeb/LaserWeb4/wiki/1.4-Installation-on-RaspberryPi

Hello, I am trying to install LaserWeb4(lw.comm-server) on my raspberry pi. The install went smoothy until executing this command "node server.js". I am receiving the following error:
[image]

Let me know if there is any other information I can provide.

cprezzi, 2018-08-27:
Ups, my fault. I pushed my changes yesterday before I fixed this error. It's fixed now.

closed 2018-08-27 by cprezzi

cprezzi, 2018-08-27:
If you used git to clone the server, you can just execute a **git pull** to fix it.

Benjin454, 2018-08-27:
Works now! Thanks!