LaserWeb/LaserWeb4 #511: System shutdown button

maxpautsch, 2018-08-28:
When running on a RaspberryPi, it is essential to shut down the system correctly in order not to damage the filesystem on the SD card.
It is inefficient to shut down the system using SSH when having such a nice GUI for the machine. I would suggest adding a shut down button to LaserWeb.

I would implement it like this:
- add a detection, if LaserWeb is running on a RaspberryPi to lw.comm-server/firstLoad
- the GUI displays a tab in between Settings and About
- on this tab, the system can be shut down using a new command to the comm-server.

In addition, there could be access to the GPIOs on the RaspberryPi on this tab. Then the user could for example control a compressor, the exhaust fans or check, if coolant is running. 

If this feature seems to be useful to you, I would implement it.

cprezzi, 2018-08-28:
A remote server shutdown could be usefull indeed, but I don't like the idea of using the RasPi GPIO's. Airassist, exhaust, cooling and such things should be handled by the machine controller and not the server. I don't want to mix the layers.

cprezzi, 2018-08-30:
Firmwares like Smoothieware allow you to configure multiple "switches" that can then be controlled by gcode. And they have options to signal status. This would be a better place to add such things.

closed 2018-12-11 by cprezzi

DIYDave, 2022-03-09:
An old thread I know. But I also lacked a way to shut down the Raspberry Pi before switching it off. That's why I made this little tool. https://github.com/DIYDave/RPIMON
Written in Python and an executable for Windows is also available.

easytarget, 2022-03-09:
That works. 

Also:
https://raspberrypi.stackexchange.com/questions/42929/how-to-safely-switch-off-the-raspberry-pi/
.. has some good instructions on how to set a GPIO pin to trigger a shutdown. Which is a better solution if you can handle the hardware. It removes the reliance on a working wifi connection.

So long as the PI has been dormant for a few seconds (green activity light has stopped) it is actually quite safe to just pull the power. 

Yes; You **can** corrupt the disk if you kill the power while a write is happening, but in practice this is very hard to do unless you do it during a system update or some other heavy operation.
- As part of my work on [OverWatch](https://github.com/easytarget/pi-overwatch) I have been monitoring disk write frequency on two machines running laserweb, and on another running 2x Octoprint. so I have some hard data on this.
- Disk write activity is literally in the one or two writes per hour range for these machines when sitting idle, Raspbian tries to minimize disk writes in order to reduce SD card wear.

That said, this is not a bad suggestion.

Laserweb cannot shut the PI down, Laserweb is running in the web browser on your local machine. `lw.comm-server` is what we run on the Pi, and lw-comm server runs entirely in memory, so it is not a source of disk writes.

To support this lw.comm-server would need to be extended to support a 'shutdown' command. With a separate change to LaserWeb to add a shutdown button. And then you get into permissions problems, do you allow 'just anyone' to send a shutdown request? do you 'password protect' it? do you allow it if a job is running? etc..

The point of the above is not to diminish the suggestion, but rather to note that this is a very low priority item from my perspective; it's more complex than it appears on the surface. Plus there are good workarounds available, eg buttons, traditional ssh, tools like @DIYDave 's, waiting for inactivity, or just leaving the pi running 24/7.

cprezzi, 2022-03-09:
I always use cases with power buttons for my Pi's like this:
https://www.okdo.com/de/p/okdo-black-power-hat-case-for-pi-4-model-b/
This way you can shutdown or start the PI with the push of a button.

easytarget, 2022-03-09:
All my Pi's are independently powered, so they run 24/7, but I'm pretty careful to SSH/shutdown them before unplugging. Most of the time.

DIYDave, 2022-03-09:
The case is nice! So everyone has their own solution. :-)
I put my RPi in a box with all the other stuff. Therefore, the software solution is ideal for me. Thank you and best regards
[image]
.