LaserWeb/LaserWeb4 #514: Some lines in svg missing

Peetz0r, 2018-09-02:
I made this svg file on http://www.makercase.com/ and edited it with inkscape 0.92.3, then tried to cut it with Laserweb 4..0.991/4.0.127, and one set of lines went missing in the resulting gcode.

[Workspace.json and svg in .zip](https://github.com/LaserWeb/LaserWeb4/files/2343679/workspace-and-svg.zip)

[image]

[image]

Peetz0r, 2018-09-02:
Possible duplicate of #442 and #446.

cprezzi, 2018-09-04:
It's known that LW could have some problems with makercase files. The problem is mostly caused by two lines on top of each other. A possible workaround is to separate the different parts (move apart).

valdasa, 2018-09-27:
same thing happens with coreldraw exports, but funniest thing that if you use same file on laserweb3 old version everything is ok

cprezzi, 2018-09-27:
Nothing funny or magic at all. LW4 uses a complete different SVG parser than LW3 used, which has many advantages but unfortunately also some disadvantages.

valdasa, 2018-09-28:
yeah but lw4 parser isn`t working for my needs, it`s super slow generating takes up to couple minutes on i7... and it generates incorrectly. So for raster i use lw4 for cutting lw3.

cprezzi, 2018-09-28:
@valdasa It's known that CorelDraw doesn't make perfect SVG files. Did you try to import the file into inkscape and save the svg from there?. That could eventually fix some of the problems.

cojarbi, 2018-09-28:
> yeah but lw4 parser isn`t working for my needs, it`s super slow generating takes up to couple minutes on i7... and it generates incorrectly. So for raster i use lw4 for cutting lw3.

Try using the segment field with a 0.5 value and try again the cut operation.

closed 2018-12-11 by cprezzi