LaserWeb/LaserWeb4 #516: Build fail on Linux

FabriceSalvaire, 2018-09-27:
We would like to rebuild LaserWeb on Windows for our hackerspace, but I cannot rebuild it on Linux ( as well as on Windows )

> npm start

```
     ERROR in ./src/lib/cam.js
     Module not found: Error: Can't resolve 'web-cam-cpp' in '/home/scratch/sources/LaserWeb4/src/lib'
     @ ./src/lib/cam.js 41:0-36
     @ ./src/lib/cam-gcode-laser-cut.js
     @ ./~/babel-loader/lib?{"presets":["react"],"plugins":["transform-es2015-destructuring","transform-es2015-parameters","transform-object-rest-spread","transform-es2015-modules-commonjs","react-hot-loader/babel"]}!./src/lib/workers/cam-lasercut.js
```

> npm run bundle-prod

```
    > laserweb@4.0.996 bundle-prod /home/scratch/sources/LaserWeb4
    > webpack -p --progress --colors

    Hash: ed9bbd65ce415149d18e
    Version: webpack 2.7.0
    Time: 26262ms
    Asset Size Chunks Chunk Names
    674f50d287a8c48dc19ba404d20fe713.eot 166 kB [emitted]
    d5763a38783660196107.worker.js 40 kB [emitted]
    93e338c2b30397c0b153.worker.js 621 kB [emitted] [big]
    93e338c2b30397c0b153.worker.js.map 761 kB [emitted]
    21421bab63cc19998d92.worker.js 1.48 MB [emitted] [big]
    21421bab63cc19998d92.worker.js.map 1.91 MB [emitted]
    3f0fa5d448bae9d3631c.worker.js 320 kB [emitted] [big]
    3f0fa5d448bae9d3631c.worker.js.map 412 kB [emitted]
    6465bc71fed0a3f02b80.worker.js 1.73 MB [emitted] [big]
    6465bc71fed0a3f02b80.worker.js.map 2.21 MB [emitted]
    f4769f9bdb7466be65088239c12046d1.eot 20.1 kB [emitted]
    d5763a38783660196107.worker.js.map 50.9 kB [emitted]
    fa2772327f55d8198301fdb8bcfc8158.woff 23.4 kB [emitted]
    448c34a56d699c29117adc64c43affeb.woff2 18 kB [emitted]
    af7ae505a9eed503f8b8e6982036873e.woff2 77.2 kB [emitted]
    fee66e712a8a08eef5805a46892932ad.woff 98 kB [emitted]
    e18bbf611f2a2e43afc071aa2f4e1512.ttf 45.4 kB [emitted]
    b06871f281fee6b241d60582ae9369b9.ttf 166 kB [emitted]
    89889688147bd7575d6327160d64e760.svg 109 kB [emitted]
    912ec66d7572ff821749319396470bde.svg 444 kB [emitted] [big]
    index.js 9.42 MB 0 [emitted] [big] main
    index.js.map 11.6 MB 0 [emitted] main
    [0] ../~/react/react.js 56 bytes {0} [built]
    [12] ../~/react-dom/index.js 59 bytes {0} [built]
    [70] ./index.js 4.92 kB {0} [built]
    [544] ../~/babel-polyfill/lib/index.js 833 bytes {0} [built]
    [545] ../~/webpack-dev-server/client?http://0.0.0.0:8080 7.93 kB {0} [built]
    [546] ../~/webpack/hot/only-dev-server.js 2.29 kB {0} [built]
    [625] ../~/babel-polyfill/~/regenerator-runtime/runtime.js 24.4 kB {0} [built]
    [676] ../~/core-js/fn/regexp/escape.js 108 bytes {0} [built]
    [917] ../~/core-js/shim.js 8.18 kB {0} [built]
    [1232] ../~/redux-logger/dist/redux-logger.js 10.5 kB {0} [built]
    [1261] ../~/strip-ansi/index.js 161 bytes {0} [built]
    [1305] ../~/webpack-dev-server/client/overlay.js 3.67 kB {0} [built]
    [1308] ../~/webpack/hot nonrecursive ^\.\/log$ 160 bytes {0} [built]
    [1309] ../~/webpack/hot/log-apply-result.js 1.02 kB {0} [built]
    [1344] multi webpack-dev-server/client?http://0.0.0.0:8080 webpack/hot/only-dev-server babel-polyfill ./index.js 64 bytes {0} [built]
    + 1330 hidden modules

    ERROR in index.js from UglifyJs
    Unexpected token: name (Icon) [./components/font-awesome.js:15,6][index.js:12324,4]
    Child worker:
    Asset Size Chunks Chunk Names
    d5763a38783660196107.worker.js 40 kB 0 [emitted] main
    d5763a38783660196107.worker.js.map 50.9 kB 0 [emitted] main
    [0] ./lib/cam-gcode-lathe.js 11.5 kB {0} [built]
    [1] ../~/babel-loader/lib?{"presets":["react"],"plugins":["transform-es2015-destructuring","transform-es2015-parameters","transform-object-rest-spread","transform-es2015-modules-commonjs","react-hot-loader/babel"]}!./lib/workers/cam-lathe.js 1.42 kB {0} [built]

[... 77 lines cut ...]
    [14] ./lib/cam-gcode-laser-cut.js 11.8 kB {0} [built]
    [15] ./lib/action2gcode/gcode-generator.js 974 bytes {0} [built]
    [16] ./lib/action2gcode/generators/default-generator.js 2.38 kB {0} [built]
    [17] ./lib/action2gcode/generators/marlin-generator.js 2.75 kB {0} [built]
    [18] ./lib/cam.js 16 kB {0} [built]
    [19] ../~/babel-loader/lib?{"presets":["react"],"plugins":["transform-es2015-destructuring","transform-es2015-parameters","transform-object-rest-spread","transform-es2015-modules-commonjs","react-hot-loader/babel"]}!./lib/workers/cam-lasercut.js 1.6 kB {0} [built]
    [20] ../~/gl-matrix/lib/gl-matrix.js 548 bytes {0} [built]
    [23] ../~/gl-matrix/lib/gl-matrix/quat2.js 22.6 kB {0} [built]
    [28] ../~/poly2tri/src/sweepcontext.js 14.8 kB {0} [built]
    [32] ../~/script-loader!../~/web-cam-cpp/web-cam-cpp.js 235 bytes {0} [built]
    + 31 hidden modules

    ERROR in 6465bc71fed0a3f02b80.worker.js from UglifyJs
    Unexpected token: name (AbstractGenerator) [./lib/action2gcode/generators/abstract-generator.js:4,6][6465bc71fed0a3f02b80.worker.js:896,4]
    npm ERR! code ELIFECYCLE
    npm ERR! errno 2
    npm ERR! laserweb@4.0.996 bundle-prod: `webpack -p --progress --colors`
    npm ERR! Exit status 2
    npm ERR!
    npm ERR! Failed at the laserweb@4.0.996 bundle-prod script.
    npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

    npm ERR! A complete log of this run can be found in:
    npm ERR! /home/fabrice/.npm/_logs/2018-09-27T13_12_35_916Z-debug.log
```

cprezzi, 2018-09-27:
"bundle-prod" is broken since long time. Just use "bundle-dev" instead.

FabriceSalvaire, 2018-09-27:
Thanks for the tip ! I could now run LaserWeb on Linux using

```
npm run bundle-dev
npm start
```

**Would be fine to update wiki/How-to-Contribute**

closed 2018-09-27 by FabriceSalvaire

myk3y, 2018-10-25:
> laserweb@4.0.996 bundle-dev /home/mike/LaserWeb4
> webpack --progress --colors

 66% building modules 933/984 modules 51 active ...e_modules/react-overlays/lib/Modal.jsKilled

npm ERR! Linux 4.14.18-sunxi
npm ERR! argv "/usr/bin/node" "/usr/bin/npm" "run" "bundle-dev"
npm ERR! node v6.14.4
npm ERR! npm  v3.10.10
npm ERR! code ELIFECYCLE
npm ERR! laserweb@4.0.996 bundle-dev: `webpack --progress --colors`
npm ERR! Exit status 137

Bundle=dev doesn't seem to work, either.