LaserWeb/LaserWeb4 #517: REST API to perform accounting

FabriceSalvaire, 2018-09-27:
Just to inform, we would like to implement an accounting feature in our hackerspace.

Basically, a REST API to log user, job time and laser on time.

FabriceSalvaire, 2018-09-27:
Just found in lw-comm-server

```
config.jobOnStart = '';
config.jobOnFinish = '';
config.jobOnAbort = '';
```

but pass any data

cprezzi, 2018-09-28:
Those are hooks to start scripts outside of lw-comm-server at certain events like start, abort or finish of a job. With a start, abort and finish script you could send the event and timestamp to your database, but I don't see a way to get a username from the frontend.

In our FabLab we are working on a RFID based machine controller that will be used for authorisation and billing. The machine can be locked/unlocked via a relais or optocoupler output. See https://github.com/fablabwinti/RFID_Machine_Controller for Details.

closed 2018-12-11 by cprezzi