LaserWeb/LaserWeb4 #518: SVG problem

Vathi2018, 2018-10-18:
How can this be?
The opening/mismatch is the start and end of lasering the line.

See test.jpg, Test 2.jpg & logo.jpg
(I also included 2 SVG files)
All made with Adobe Illustrator and exported the right way.
Running everything on a Mac.

I asked around on the Google+ community but no solution.
Is it a software error or has it something to do with my machine?
I tried different speeds and sizes, all the same result.

Just thought to bring it to your attention, I will look further for answers.

Great Programm, great work you guys did!

Kind regards,
Hans

[image]

[image]

[image]

[SVG.zip](https://github.com/LaserWeb/LaserWeb4/files/2491596/SVG.zip)

tatarize, 2018-12-09:
The relevant gcode says:

```
; Pass 0 Path 12
G0 X18.81 Y15.11
M4
G1 X18.80 Y15.18 S1000.00 F2000
G1 X18.80 Y15.24
... (skipping a bunch of shape)
G1 X18.81 Y15.05
G1 X18.81 Y15.11
M5
```

It turns on the laser and powers up at (18.80, 15.11) and powers back down at (18.80, 15.11) this means it is at the same spot as far as the gcode is concerned.

If I had to guess, it's a position rounding error. The curve it's drawing is a bunch of line segments and those line segments all have a bunch of fractional amounts all biased towards one end.  But, the error is not in the gcode.

tatarize, 2018-12-09:
If you are, for example, using K40 Whisperer, it will have this compounded sub-fractional rounding bug in the egv.py file:

```
            for i in range(1,len(ecoords_in)):
                e0,e1,e2                = self.ecoord_adj(ecoords_in[i]  ,scale,FlipXoffset)
                ...
                dx = e0 - lastx
                dy = e1 - lasty
                ...
                    if laser:
                        ...
                        self.make_cut_line(dx,dy,Spindle)
                    else:
                            ...
                            self.rapid_move_fast(dx,dy)
                lastx     = e0
                lasty     = e1
                last_loop = e2
```

With the ecoord_adj element critically reading.

```
    def ecoord_adj(self,ecoords_adj_in,scale,FlipXoffset):
        if FlipXoffset > 0:
            e0 = int(round((FlipXoffset-ecoords_adj_in[0])*scale,0))
        else:
            e0 = int(round(ecoords_adj_in[0]*scale,0))
        e1 = int(round(ecoords_adj_in[1]*scale,0))
        e2 = ecoords_adj_in[2]
        return e0,e1,e2
```

The bug here is really really obscure. But, basically, when you are converting coordinate systems from absolute to relative like this, even the tiny fractional throwaway elements are actually relevant and *must* be accounted for. The problem here is that it rounds the values, then uses those rounded value numbers directly as the "lastx" and "lasty" values. Imagine you put absolute value coordinates that move by 0.333333 units each time, and it rounds to 1 unit. If you do this 10,000 times in the absolute coords you have moved 3,333 units. However, if you rounded the dx each time, you would *always* get 0. You would never actually get a dx or dy above 0. So you would travel 0 units rather than correctly travelling 3333 units. Properly you actually need to travel 0,0,1,0,0,1,0,0,1... units by manually compounding sub-fractional elements you would otherwise round off.

It actually needs to be written with a local variable that is the compound of the rounded values.

```
        xx = 0
        yy = 0
        for coords in ecoords_in:
            x = coords[0]
            y = coords[1] # apply any required scaling here.
            dx = int(round(x - xx)) 
            dy = int(round(y - yy))  # or whatever rounding ops are needed.
            xx += dx;
            yy += dy;
            # then use the values of the ecoords.
```

Keep in mind though, this bug will only matter if the compounding error within the segments occurs constantly on one side. This basically only happens on curves like that. Usually on par the subfractional rounding error is pretty much random and performs a random walk, and usually can't compound to anything noticeable, however with arc-curves like that they really can start to add up and result in an error between the relative-coordinates and absolute-coordinates that is noticable.  That's why it happens on curves but not on other elements, on other elements any errors from this bug will be a wash. But, it's a bug with @scorchworks and not LaserWeb. Though, likewise it's an obscure bug type and whatever software you're using that converts the gcode absolute coords into relative coords for the xy-plotter of the laser cutter is making this same error highlighted here.

I assume that's K40 Whisperer but I don't actually know. I'd be happy to notify them of the bug, even if it took skimming through for the one part where this happens and noting that it does. I've diagnosed this error before in Embroidery software that likewise does absolute coords to relative coords for a physical xy-plotter, and suddenly got weird when it tried to approximate curves ( Embroidermodder/Embroidermodder#113 ), ( inkstitch/inkstitch#231 ).

Vathi2018, 2018-12-09:
Thank you for the reply.

In the meanwhile I found a way to work around it.
I gave the first D a other color so the route laser takes to burn is different now.
Shapes are correct now :-)

closed 2018-12-09 by Vathi2018

tatarize, 2018-12-09:
The K40 output actually seems correct for the file anyway. Likely something elsewhere or something else you're using. Cuts from 347,-1172 around the arc and back to 347,-1172 so it must be something else. Could even just be a registration error on that run or something.

3565  dir  -418   -17  :  347   -1172
3566  cut  -1   3  :  346   -1169
3567  cut  0   2  :  346   -1167
...
3938  cut  0   3  :  347   -1174
3939  cut  0   2  :  347   -1172
3940  dir  -3   0  :  344   -1172