LaserWeb/LaserWeb4 #522: 4.0.996 Server installation on Raspberry

cotecmania, 2018-10-29:
Hello,

when i install on Raspberry like mentioned in documentation I only get version 4.0.990 installed.
(git clone https://github.com/LaserWeb/lw.comm-server.git)
How can I install 4.0.996 ?

Thanks
Joe

cprezzi, 2018-10-30:
Try to delete folder `node-modules`, then do a `git pull` and `npm install`.

JoeCotty, 2018-10-30:
> 
> 
> Do you mean you get v4.0.990 when you call the website at server:8000?

Yes.

> 
> 
> Try to delete folder `node-modules`, then do a `git pull` and `npm install`.

Deleted folder but then >git pull says : Already up-to-date

What now ?

cprezzi, 2018-10-31:
Ok, now you need to run `npm install`

JoeCotty, 2018-11-04:
npm install does not work. There are error messages and missing binaries

So i did : 
>sudo npm install serialport --unsafe-perm --build-from-source

which runs through butt tells in the ende that there are vulnerabilities and i shoul run 
>npm audit fix

running this says i should use  
>npm audit fix --force

After that i started the server but frontend still says  4.0.990

mzeile, 2018-11-09:
Same here. Removed my whole lw.comm-server directory to get a clean installation. Also ended up with 4.0.990 only.

cprezzi, 2018-11-12:
I advise to use the binary installers for the frontend (and connect to the raspi server from there), because only the binary installers get the newest frontend version. The embedded frontend in lw-comm-server is allways some version behind.