LaserWeb/LaserWeb4 #524: Superimposed graphic items are not cutted

FabriceSalvaire, 2018-11-14:
When we make a dxf using Fusion, especially for array patterns, we can have an item twice in the dxf. Just because user selected the array and unfortunately the model item behind the first repeated item.

Unfortunately LW doesn't cut theses items, nor warn user.

I think this case could be fixed easily since theses items have identical properties : same polylines / circles.

IMHO, the JavaScript design choice, make difficult the use of advanced algorithms to improve this.  Like a clearance test.

cprezzi, 2018-11-15:
I'm sorry that LaserWeb is not the "professional" solution with all your preferred safety and convenience features, but it's free and open source. So you can just make a fork and add as many features as you like. We would be happy to accept your pull requests.

Or you can ask @tbfleming (our master frontend designer) if he is keen on developing your preffered features for some beer money ;)

FabriceSalvaire, 2018-11-16:
As an open source developer I totally agree with you and we (the hackerspace) would like to contribute to LW.

But unfortunately the electron framework is repulsive for a lot of us.

cprezzi, 2018-11-19:
I suggest to start without electron and just use LaserWeb branch dev-es6 and lw.comm-server master.
The frontend is a pure browser application based on JS, React and Redux that uses websockets to communicate with the backend. The backend (server) is a NodeJS application that bridges between websocket API and machine USB serial port.

The frontend uses open source libraries to parse DXF or SVG files into an internal structure (I'm not the frontend specialist, I care about the backend). Those libraries are not perfect. If you need more help to get into frontend development, please ask @tbfleming.