LaserWeb/LaserWeb4 #525: Out of bound issue

FabriceSalvaire, 2018-11-14:
Machine is a laser cutter drived by a SmoothieBoard.

* when a path has x_min = 0, i.e. it touches an axis, ( cut mode set to cut outside ), the laser cut the mounting material used to set the zero offset.
* **[SAFETY and BUG]** when a segment has a negative coordinate, the laser don't move and burn in place, at the negative position

cprezzi, 2018-11-14:
The cam operations will not move any parts automatically, you need to move the parts yourself away from the axes to get enough free space for your tool (mill, laser...).

FabriceSalvaire, 2018-11-14:
It means there is any clearance test for a dangerous machine used by lambda (basic training) users.

I would expect a dilation of the path and an inclusion test with the dual of the work area. Where the clearance is set as a configuration parameter.

Well something similar to the pcb manufacturing in the industry.

cprezzi, 2018-11-15:
Valid axes ranges is usually checked by the machine controller, either by limit switches or soft limits, which generates an alarm condition when the gcode drives into the ends. A correctly configured firmware will stopp the laser in such a case. I don't see any dangerous situation here.

cprezzi, 2018-11-15:
Why don't you just set the origin somewhare away from the limits?

cprezzi, 2018-11-15:
The problem is, LaserWeb is made for many different kinds of machines. On a mill for example, the work origin is often set on features (like corner oder center of hole) of the workpiece and working in nagative coordinates is needed. Therefore we can not block negative coordinates. The operator is responsible to place the piece inside the workspace and check that the tool (head) has enough space around it.

FabriceSalvaire, 2018-11-16:
We have to investigate further, maybe it is a SmoothieBoard issue.

arthurwolf, 2018-11-16:
If you want smoothieboard to do this, you want to look at
http://www.smoothieware.org/endstops , specifically the section about "soft
endstops"

cprezzi, 2018-11-19:
@FabriceSalvaire I don't think this is a smoothieware "issue", it's just a configuration thing of the soft limits. I also understand that you would prefer to have an automatic part placement/shift function (which theoretically is possible for lasercutters), but LaserWeb is made for many different machinetypes and therefore it's complicated to implement features for just one machinetype without creating problems on others.

Thanks @arthurwolf for helping out.

cprezzi, 2018-11-19:
@FabriceSalvaire Just for your reference, here is the process I usualy do when lasercutting:

- Load the SVG/DXF file
- Move the parts with the LaserWeb placement popup to 0,0 (if not already there).
- Create the Operations
- Create Gcode
- Connect to machine
- Home the machine (this sets machine origin)
- Move the head to the location where you want the bottom left corner of the part (depending on placement of material)
- Set Zero X and Y (this sets workpiece origin)
- Start the job