LaserWeb/LaserWeb4 #527: Timing/Speed/Interface ? problem

JoeCotty, 2018-11-22:
Hello,

i have problem to engrave the attached SVG or gcode. It seems that my laser cutter drives to each point and stays there for a second. So the result is black burned wood.
It makes no difference if i use the windows frontend 4.0.996 or the web frontend 4.0.990
Laserweb server is running on Raspy 3.
Firmware is grbl1.1f.
This is the first SVG which does not work as expected.
Easyer models which have less points (tabbed box, circles, squares) are working as expected.

Also if i use LaserGRBL the same SVG works perfekt and is engraved very fast and beautiful.

[ornament.txt](https://github.com/LaserWeb/LaserWeb4/files/2607441/ornament.txt)
[Laserweb-Workspace.json.txt](https://github.com/LaserWeb/LaserWeb4/files/2607445/Laserweb-Workspace.json.txt)
[ornament.svg.txt](https://github.com/LaserWeb/LaserWeb4/files/2607450/ornament.svg.txt)

Please help because I can not always connect directly with LaserGRBL and I need the LaserWebServer.
Thanks
Joe

cprezzi, 2018-11-22:
Use M4 instead of M3 to activate the laser. M4 sets the laser PWM in relation to motion speed (adaptive laser power), which solves your problem. Check Settings / Gcode / Tool on. (M4 is only available on grbl!)

JoeCotty, 2018-11-23:
Problem solved. Rebooted Raspy !

I detected also a big delay between clicking the manual JOG-Button and moving the laser.
This wasn't in the beginning, so i rebooted the server.
Now everything runs fast and fine like before.

The bad thing is that i do not know the root cause and this effect can repeat and comes slowly, so that effects like bad engraving will also come slowly ...